NLRWindSystems/WEIS #304: Open MPI dependency broken with temporary resolution

yonghoonlee, 2024-07-24:
## Description
Open MPI creates segmentation faults on Linux machines.

### Steps to reproduce issue

1. Follow the WEIS develop branch installation instruction, then mpi=1.0-openmpi version is installed along with mpi4py=3.1.6 version.

2. Run any MPI job, it will fail. More (generalized) information found on [this page](https://github.com/conda-forge/openmpi-feedstock/issues/171)

3. Temporary resolution to this issue is to install `mpich` version of `mpi` instead of `openmpi`. When installing the WEIS, install specific build of mpi:
```
conda install -y petsc4py mpi4py mpi=1.0=mpich pyoptsparse     # (Mac / Linux only)
```
instead of running
```
conda install -y petsc4py mpi4py pyoptsparse     # (Mac / Linux only)
```
to forcefully install `mpich` version of `mpi`.

### Current behavior
```
[log02:599094:0:599094] Caught signal 11 (Segmentation fault: address not mapped to object at address 0x3ff00000001)
[log02:599097:0:599097] Caught signal 11 (Segmentation fault: address not mapped to object at address 0x3ff00000001)
[log02:599095:0:599095] Caught signal 11 (Segmentation fault: address not mapped to object at address 0x3ff00000001)
[log02:599096:0:599096] Caught signal 11 (Segmentation fault: address not mapped to object at address 0x3ff00000001)
```

### Expected behavior

### Code versions

yonghoonlee, 2024-07-24:
At this moment, the issue can be temporarily circumvented by specifying `mpich` as described above. I hope the dependency issue can be resolved from the condo-forge feedstock level soon.

dzalkind, 2024-07-31:
@yonghoonlee, is this happening on Kestrel or another linux machine?

yonghoonlee, 2024-07-31:
@dzalkind Yes, it happened in all my Linux machines as well as HPC systems I am currently using, including Kestrel and UofM HPC. When you specify mpich then it works fine. Otherwise, conda automatically select MPI dependencies, and it could be either openmpi or mpich. If openmpi is automatically selected, then the problem persists.

dzalkind, 2024-08-06:
Solution from @yonghoonlee 

`conda install -y petsc4py mpi4py mpich pyoptsparse     # (Mac / Linux only)`

(also install mpich)

closed 2025-02-14 by dzalkind