NLRWindSystems/WEIS #310: No module named 'rosco'

mlidGWT, 2024-08-22:
## Description
Hello friends,

It's been a while since I've run WEIS (have you missed me?), and I came back to it today on my M1/2 Mac Mini. When I last ran WEIS on this machine I don't recall having any issues, but I know a lot has been changing in WEIS over the past months. 

I initially tried running Example 08_OLAF, and got the error:

> Traceback (most recent call last):
  File "/Users/meglidrbauch/Software/WEIS/examples/08_OLAF/weis_driver.py", line 5, in <module>
    from weis.glue_code.runWEIS     import run_weis
  File "/Users/meglidrbauch/Software/WEIS/weis/glue_code/runWEIS.py", line 4, in <module>
    from weis.glue_code.gc_LoadInputs     import WindTurbineOntologyPythonWEIS
  File "/Users/meglidrbauch/Software/WEIS/weis/glue_code/gc_LoadInputs.py", line 5, in <module>
    from rosco import discon_lib_path
ModuleNotFoundError: No module named 'rosco'

Line 5 of gc_LoadInputs.py simply says `from rosco import discon_lib_path` so I don't know where to go from there.

I recloned the `develop` branch after renaming my existing WEIS folder and retried running example 08 from the new clone but got the same error. Unsure if it was OLAF-specific, I tried running Example 06 and got the same error as above. 

From some help @rafmudaf  gave our intern over the summer, I think there is a path/pointer I need to change in some file to fix the issue, but I forget where and what to change it to. Posting here as an issue in case anyone else runs into the same thing, or in case a more permanent fix can be made.

Thanks in advance for the help!

gbarter, 2024-08-25:
Hi Meg- If you haven't used WEIS in a number of months, I would suggest either updating everything in your conda environment and then re-installing, or starting a new environment from scratch.  All of the key codes have gone through many updates, libraries and files have moved, etc.  That is why you are seeing that error.

mlidGWT, 2024-08-28:
Hi Garrett, 

Making a new environment fixed the problem, thank you for pointing that out, I will add that to my troubleshooting list going forward!

Best,
Meg

closed 2024-08-28 by mlidGWT