NLRWindSystems/WEIS #311: OLAF changes hub height?

mlidGWT, 2024-08-28:
## Description
Hello, 

I recently ran a test to familiarize myself with OLAF. With a fresh conda environment and clone of the WEIS repository `develop` branch, I modified the `modeling_options.yaml` from Example 06 to run DLC 6.1, first with a BEM baseline (no changes to the yaml beyond the output file prefix and folder name, and the change to DLC 6.1) and then duplicated to an OLAF version, where I copied in the `Level3>AeroDyn` section from the Example 08 OLAF `modeling_options.yaml` file without changing any values. I ran WEIS in both instances pointing to the same IEA 15MW RWT geometry file and analysis_options.yaml (both from Example 06, though I removed the optimizations in `analysis_options`). 

When the simulations completed, I started by looking at the wind profiles in PyDatView and was confused to see that they were different, since the case matrix files indicated the same wind seed had been used. With some further digging, I saw that in the 'Turbine/Model Specifications' section of the wind `*.sum` files, the Hub Height was correctly listed as 150m for the baseline case, but somehow changed to 450m for the OLAF case, with a correspondingly large wind field. 

I have not knowingly changed the hub height for the OLAF case, and the `InflowWind.dat` files are identical between the two scenarios beyond the filename prefixes specified in the `modeling_options_*.yamls`, both with all references to height indicating 150m. Have I unwittingly changed the hub height through the copied OLAF parameters? None of the parameter names indicates it should modify hub height, which is confirmed by the parameter descriptions [here.](https://openfast.readthedocs.io/en/dev/source/user/aerodyn-olaf/InputFiles.html)

Please advise! I am attaching a zip folder, [OLAF_TestsZip.zip](https://github.com/user-attachments/files/16784938/OLAF_TestsZip.zip) that includes all relevant files and a brief Readme as to the contents. Let me know if additional information is needed.

### Steps to reproduce issue
See linked zip folder with files used to run the simulations and WEIS's outputs.

ptrbortolotti, 2024-08-28:
hello @mlidGWT, you've run into an issue that we had in the past where OLAF was crashing when the wake was hitting the ground. to go around this weakness of OpenFAST, we artificially increased the hub height. at the time we were working with a stiff tower and no shear, and so we could do this. I am not sure anymore if this bug in OpenFAST has been fixed. could you please try commenting out https://github.com/WISDEM/WEIS/blob/03b0c1fb5716bb9f2593d7a43833f3170cc7d000/weis/aeroelasticse/openmdao_openfast.py#L1810 and run again?

mlidGWT, 2024-08-28:
Hi @ptrbortolotti  thanks for the quick response! I commented out the whole `if` statement around the line you indicated, as TurbSim choked with the hub height being less than half the grid height (`dlc_generator.cases[i_case].GridHeight *= 3.` remained before I commented out the whole statement). With that change, the simulation completed and produced an out file with results, so either my simulation scenario didn't trigger the OpenFAST bug or it has been fixed.

ptrbortolotti, 2024-08-28:
@andrew-platt will know if this bug has been fixed in OpenFAST v3.5.3

ptrbortolotti, 2024-08-28:
and thank you @mlidGWT !

andrew-platt, 2024-08-29:
@mlidGWT, which version of OpenFAST are you using?  In checking the OLAF code, I see a few changes have been made to the wake reflection in the `dev` branch from a PR that has not made it into a release yet (#1791).

mlidGWT, 2024-08-30:
> @mlidGWT, which version of OpenFAST are you using? In checking the OLAF code, I see a few changes have been made to the wake reflection in the `dev` branch from a PR that has not made it into a release yet (#1791).

Hi Andy, according to my conda environment, I am running OpenFAST version 3.5.3

dzalkind, 2024-10-02:
This issue should be fixed by @ptrbortolotti 's fix in #313

closed 2024-10-02 by dzalkind