NLRWindSystems/WEIS #312: automatic working directory prepend failure

cfrontin, 2024-08-28:
## Description

Prepend of `log_opt.sql` location is failing, resulting in a IO FileNotFound error.

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. follow instructions under `Set up on HPC` on [this page](https://github.com/sora-ryu/WEIS/blob/develop/docs/how_weis_viz_works.rst)
2. run on case in [this working directory](/kfs2/projects/weis/cfrontin/software/weis/examples/17_IEA22_Optimization/17_IEA22_OptStudies/of_COBYLA) following instructions with [analysis_options_of_ptfm_opt.yaml](/kfs2/projects/weis/cfrontin/software/weis/examples/17_IEA22_Optimization/analysis_options_of_ptfm_opt.yaml) and [modeling_options_openfast.yaml](/kfs2/projects/weis/cfrontin/software/weis/examples/17_IEA22_Optimization/modeling_options_openfast.yaml)
3. Open optimization pane in viz tool

### Current behavior

File not found causes optimization pane to error out.

### Expected behavior

Sweet, sweet plots.

cfrontin, 2024-08-28:
I didn't realize at first, but it also is failing to open the OpenFAST tab as well, with a different error:
```
Traceback (most recent call last):
  File "/projects/weis/cfrontin/conda-envs/weis-env/lib/python3.12/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/projects/weis/cfrontin/conda-envs/weis-env/lib/python3.12/site-packages/flask/app.py", line 882, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/projects/weis/cfrontin/conda-envs/weis-env/lib/python3.12/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/projects/weis/cfrontin/conda-envs/weis-env/lib/python3.12/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/projects/weis/cfrontin/conda-envs/weis-env/lib/python3.12/site-packages/dash/dash.py", line 1373, in dispatch
    ctx.run(
  File "/projects/weis/cfrontin/conda-envs/weis-env/lib/python3.12/site-packages/dash/_callback.py", line 465, in add_context
    output_value = _invoke_callback(func, *func_args, **func_kwargs)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/projects/weis/cfrontin/conda-envs/weis-env/lib/python3.12/site-packages/dash/_callback.py", line 40, in _invoke_callback
    return func(*args, **kwargs)  # %% callback invoked %%
           ^^^^^^^^^^^^^^^^^^^^^
  File "/kfs2/projects/weis/cfrontin/software/weis/weis/visualization/appServer/app/pages/visualize_openfast.py", line 93, in define_graph_cfg_layout
    channels = sorted(df1['file1'][0].keys())
                      ~~~~~~~~~~~~^^^
IndexError: list index out of range
```

sora-ryu, 2024-08-28:
For the OpenFAST tab issue, have you added the openfast file path on the yaml file? We need at least one of them to work properly.

sora-ryu, 2024-08-28:
After reproducing the steps, generated yaml file has some unusual characters. Could you check on this @mayankchetan ?
[image]

ptrbortolotti, 2025-03-25:
I believe OpenMDAO v3.38.0 has now fixed this issue, see https://github.com/OpenMDAO/OpenMDAO/issues/3453 and https://github.com/WISDEM/WISDEM/pull/619, or is this something else?

closed 2025-04-01 by ptrbortolotti