NLRWindSystems/WEIS #314: Questions regarding WEIS (Level 1) Geometry Input Format

hymdrj, 2024-09-04:
Hello, 

I have some questions about WEIS (Level 1) geometry input 

1. Mooring:

- When defining the line type as "chain," are the material properties of the chain pre-defined in the program?

- According to the documentation, there's a "custom" option among the mooring types. It seems that users can input user-defined mooring stiffness values when using this option. Could you provide the input format required for this?

- I encountered an error when trying to model six mooring lines. Is it not possible to model six mooring lines within WEIS?

2. Member:

- In the Member section, the outer shape can be modeled as either circular or polygonal. Could you provide the input format required when modeling a polygonal shape? The documentation mentions the necessary values for polygonal modeling, but the exact format is not provided.

- Additionally, if a horizontal member is created and the axial_joint grid is set to 0.5, will the joint be created at the midpoint of the horizontal member?

3. Mass:

- There are two types of mass mentioned: floating mass and raft platform mass. What is the difference between these two?

- It seems like the FOWT mass is calculated based on the member base. Is there any documentation or resources where this is detailed?

Thank you for reading.

dzalkind, 2024-09-04:
1. We use the chain [properties defined by the MoorDyn tool](https://github.com/NREL/MoorPy/blob/018e976d63f2b5d6c8e1ba26f87d5221590886e3/moorpy/MoorProps_default.yaml#L43).  Input requirements are defined in the schema, [here are the mooring line properties](https://github.com/WISDEM/WISDEM/blob/0b3ac37b00778d7366de2084043c979183e2fc8c/wisdem/inputs/geometry_schema.yaml#L2040), inherited from WISDEM.  Six lines should be possible.  Please feel free to share your error.

2. Right now, only circular and rectangular members are supported in WISDEM. The [input requirements can be found here](https://github.com/WISDEM/WISDEM/blob/0b3ac37b00778d7366de2084043c979183e2fc8c/wisdem/inputs/geometry_schema.yaml#L1610).  Your understanding of axial joints is correct.

3. These masses should be very close.  One is calculated by [floatingse](https://wisdem.readthedocs.io/en/master/wisdem/floatingse/index.html), and one by [RAFT](https://openraft.readthedocs.io/en/latest/).

hymdrj, 2024-09-06:
First of all, thank you for your kind and detailed response

I encountered the following error when trying to model with 6 mooring lines:
ValueError: 'wisdem.wt.wt_init.mooring.moorjoint' <class MooringJoints>: Failed to set value of 'anchor_nodes': could not broadcast input array from shape (3,3) into shape (6,3).
Could you suggest a solution for this issue?

Regarding the RAFT platform mass in WEIS:
Is the platform mass in RAFT excluding any specific component from the total mass (displacement)?
According to the NREL report, it seems that the mass of the tower and RNA are excluded. Additionally, I noticed that the platform mass changes slightly every time I adjust the anchor position. Does this happen because the initial tension is adjusted when the anchor position is changed, leading to a modification in the ballast amount on the platform, and hence, changing the platform mass?

In the NREL WEIS Volturn example, the platform mass calculated by RAFT was 15,607 tons. However, in the paper "An Open-Source Frequency-Domain Model for Floating Wind Turbine Design Optimization," the Volturn platform mass calculated by RAFT was stated as 17,854 tons. Why is there a discrepancy between the platform mass calculated in WEIS and the mass calculated directly in RAFT? Could this be related to the issue mentioned in the before question?

In WEIS, the user defines the wave scatter diagram and the DLC conditions (such as 1.6, 6.1) for optimization. Is it possible to define custom wind and wave conditions for analysis without relying on DLCs? It seems like there should be a way, but I haven't been able to find it yet.

dzalkind, 2024-09-06:
It seems like you may not have added additional mooring nodes with anchor points to the geometry input.  If you share the mooring section of your geometry input, we can help you debug.

Yes, I believe that is what's happening when you change the anchor position.  Water ballast is part of the total platform mass calculated in RAFT. 

It could be a different in the model.  Sometimes end caps and bulkheads are defined differently in these models as well.  It might be difficult to track specific differences.

Right now, you must use some DLC.  However, you are free to change the severe sea state in the `metocean_conditions` portion of the modeling input, or you can set your own `wave_height` and `wave_period` within each DLC.

hymdrj, 2024-09-09:
I am encountering a persistent error with the mooring input
Failed to set value of 'anchor_nodes': could not broadcast input array from shape (3,3) into shape (6,3).

    mooring:
        nodes:
            - name: line1_anchor
              node_type: fixed
              joint: anchor1
              anchor_type: drag_embedment
            - name: line2_anchor
              node_type: fixed
              joint: anchor2
              anchor_type: drag_embedment
            - name: line3_anchor
              node_type: fixed
              joint: anchor3
              anchor_type: drag_embedment
              
            - name: line4_anchor
              node_type: fixed
              joint: anchor4
              anchor_type: drag_embedment
            - name: line5_anchor
              node_type: fixed
              joint: anchor5
              anchor_type: drag_embedment
            - name: line6_anchor
              node_type: fixed
              joint: anchor6
              anchor_type: drag_embedment

            - name: line1_vessel
              node_type: vessel
              joint: col1_fairlead
              fairlead_type: rigid
            - name: line2_vessel
              node_type: vessel
              joint: col2_fairlead
              fairlead_type: rigid
            - name: line3_vessel
              joint: col3_fairlead
              node_type: vessel
              fairlead_type: rigid
              
            - name: line4_vessel
              node_type: vessel
              joint: col4_fairlead
              fairlead_type: rigid
            - name: line5_vessel
              node_type: vessel
              joint: col5_fairlead
              fairlead_type: rigid
            - name: line6_vessel
              joint: col6_fairlead
              node_type: vessel
              fairlead_type: rigid

        lines:
            - name: line1
              node1: line1_anchor
              node2: line1_vessel
              line_type: main
              unstretched_length: 850.0
            - name: line2
              node1: line2_anchor
              node2: line2_vessel
              line_type: main
              unstretched_length: 850.0
            - name: line3
              node1: line3_anchor
              node2: line3_vessel
              line_type: main
              unstretched_length: 850.0
              
            - name: line4
              node1: line4_anchor
              node2: line4_vessel
              line_type: main
              unstretched_length: 850.0
            - name: line5
              node1: line5_anchor
              node2: line5_vessel
              line_type: main
              unstretched_length: 850.0
            - name: line6
              node1: line6_anchor
              node2: line6_vessel
              line_type: main
              unstretched_length: 850.0
              
  Also, i have some questions about DLC 
  
1. In DLC 1.6, when I input the wind speeds as [4., 6., 8., 10., 12., 14., 16., 18., 20., 22., 24.], the wave heights and periods I enter are linearly interpolated and used in the RAFT simulation. Could you explain why this method is considered in DLC 1.6?

2. "How can I ensure that the metocean conditions I input are used directly in the RAFT simulation, without any additional post-processing as I mentioned before?"

3. "Why aren’t aerodynamic forces considered in DLC 6.1? Based on my understanding of the RAFT theory, it seems that aerodynamic forces are only calculated when turbine control is included, which could be why they are excluded in DLC 6.1. Is this correct?"

dzalkind, 2024-10-02:
Hi,

Sorry for the delay.  I suspect the issue is that some of your anchors are at the same location.  I'm not able to reproduce your issue with the snippet selected.  If you send your full geometry input to dzalkind (at) nrel (dot) gov, I can take a closer look.

1. Many users have severe sea state information as a function of wind speed.  
2. I believe I answered this one in a previous response.
3. You are correct here.

Best, Dan

closed 2024-11-08 by dzalkind