NLRWindSystems/WEIS #316: Error when running WEIS examples following installation

Ajaru97, 2024-10-02:
Hi,

I have attempted to install WEIS using the instructions provided but I am getting the following error when trying to run the examples.

## Description
It seems that when I run the examples, there is an error within TurbSim within the Miniforge environment as I receive this error:

`OSError: [Errno 8] Exec format error: '/home/ajaru/miniforge3/envs/weis-env/bin/turbsim'`

### Steps to reproduce issue
1. Install on WSL using the instructions provided.
2. Run examples e.g. test_examples_skinny.py

### Current behavior
Examples such as 06_IEA-15-240-RWT/weis_driver_TMDs,  06_IEA-15-240-RWT/weis_driver_monopile and  02_run_openfast_cases/weis_driver_rosco_opt return the error when trying to run them. This is the full traceback that I receive:

```
FAIL: test_all_scripts (__main__.TestSkinnyExamples.test_all_scripts) [Running: 02_run_openfast_cases/weis_driver_rosco_opt] (i=0)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/system.py", line 2738, in _call_user_function
    yield
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/explicitcomponent.py", line 268, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "/mnt/c/Users/ajaru/WEIS/weis/aeroelasticse/openmdao_openfast.py", line 617, in compute
    summary_stats, extreme_table, DELs, Damage, case_list, case_name, chan_time, dlc_generator  = self.run_FAST(inputs, discrete_inputs, fst_vt)
                                                                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/ajaru/WEIS/weis/aeroelasticse/openmdao_openfast.py", line 1848, in run_FAST
    WindFile_type[i_case] , WindFile_name[i_case] = generate_wind_files(
                                                    ^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/ajaru/WEIS/weis/aeroelasticse/turbsim_util.py", line 260, in generate_wind_files
    wrapper.execute()
  File "/mnt/c/Users/ajaru/WEIS/weis/aeroelasticse/turbsim_util.py", line 220, in execute
    subprocess.call(exec_string)
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 389, in call
    with Popen(*popenargs, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 1955, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/ajaru/miniforge3/envs/weis-env/bin/turbsim'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/c/Users/ajaru/WEIS/weis/test/test_examples_skinny.py", line 22, in test_all_scripts
    execute_script(s)
  File "/mnt/c/Users/ajaru/WEIS/weis/test/utils.py", line 35, in execute_script
    spec.loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/mnt/c/Users/ajaru/WEIS/examples/02_run_openfast_cases/weis_driver_rosco_opt.py", line 16, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/ajaru/WEIS/weis/glue_code/runWEIS.py", line 211, in run_weis
    wt_opt.run_driver()
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/problem.py", line 749, in run_driver
    return driver._run()
           ^^^^^^^^^^^^^
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/driver.py", line 821, in _run
    self.result.success = not self.run()
                              ^^^^^^^^^^
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/wisdem/optimization_drivers/nlopt_driver.py", line 233, in run
    model.run_solve_nonlinear()
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/system.py", line 4749, in run_solve_nonlinear
    self._solve_nonlinear()
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/group.py", line 3529, in _solve_nonlinear
    self._nonlinear_solver._solve_with_cache_check()
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/solvers/nonlinear/nonlinear_runonce.py", line 26, in _solve_with_cache_check
    self.solve()  # don't use caching
    ^^^^^^^^^^^^
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/solvers/nonlinear/nonlinear_runonce.py", line 45, in solve
    self._gs_iter()
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/solvers/solver.py", line 897, in _gs_iter
    subsys._solve_nonlinear()
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/explicitcomponent.py", line 297, in _solve_nonlinear
    self._compute_wrapper()
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/explicitcomponent.py", line 251, in _compute_wrapper
    with self._call_user_function('compute'):
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/contextlib.py", line 158, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/system.py", line 2744, in _call_user_function
    raise err_type(
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/system.py", line 2738, in _call_user_function
    yield
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/site-packages/openmdao/core/explicitcomponent.py", line 268, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "/mnt/c/Users/ajaru/WEIS/weis/aeroelasticse/openmdao_openfast.py", line 617, in compute
    summary_stats, extreme_table, DELs, Damage, case_list, case_name, chan_time, dlc_generator  = self.run_FAST(inputs, discrete_inputs, fst_vt)
                                                                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/ajaru/WEIS/weis/aeroelasticse/openmdao_openfast.py", line 1848, in run_FAST
    WindFile_type[i_case] , WindFile_name[i_case] = generate_wind_files(
                                                    ^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/ajaru/WEIS/weis/aeroelasticse/turbsim_util.py", line 260, in generate_wind_files
    wrapper.execute()
  File "/mnt/c/Users/ajaru/WEIS/weis/aeroelasticse/turbsim_util.py", line 220, in execute
    subprocess.call(exec_string)
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 389, in call
    with Popen(*popenargs, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/ajaru/miniforge3/envs/weis-env/lib/python3.11/subprocess.py", line 1955, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: 'aeroelastic' <class FASTLoadCases>: Error calling compute(), [Errno 8] Exec format error: '/home/ajaru/miniforge3/envs/weis-env/bin/turbsim'

```
### Expected behavior
I would expect for WEIS to be able to run the examples.

### Code versions
- Python 3.12

Any help or suggestions would be appreciated. Perhaps I didn't install everything correctly?

Thanks,

Andrew

dzalkind, 2024-10-02:
I'm not sure what has gone wrong here, but let's take a close look at `/home/ajaru/miniforge3/envs/weis-env/bin/turbsim`

Can you execute that from the command line in WSL?

Ajaru97, 2024-10-07:
Hi Daniel,

Apologies, I'm not sure what you mean by this. Do I need to change my directory to this or execute the turbsim file? Please could you provide more instruction? Sorry again for my ignorance.

dzalkind, 2024-10-07:
You shouldn't need to change directories.  If you copy `/home/ajaru/miniforge3/envs/weis-env/bin/turbsim` into your WSL command line and execute it by hitting enter, what happens?

Ajaru97, 2024-10-07:
This is the error that is gives:

`-bash: /home/ajaru/miniforge3/envs/weis-env/bin/turbsim: No such file or directory`

dzalkind, 2024-10-07:
This `turbsim` exe should have been included when we installed the conda environment.  

What happens when you do `conda install -y openfast` ?

Ajaru97, 2024-10-08:
It says that it is already installed:

`# All requested packages already installed.`

Ajaru97, 2024-10-08:
Hi Daniel,

Just to add to this, I get this error when I try to run the turbsim binary within the conda environment.

```
(weis-env) root@DTU-5CG4332LZQ:~/miniforge3/envs/weis-env/bin# ./turbsim
-bash: ./turbsim: cannot execute binary file: Exec format error
```

dzalkind, 2024-10-08:
Hi,

A google search of that error seems to indicate that this message could arise if your WSL is 32-bit and you are trying to run a 64-bit executable.  Or perhaps your conda installation was in 32-bit and your `turbsim` executable is 64-bit.

I believe everything in conda is 64-bit, but don't quote me on that.  WEIS should also be installable and runnable on native Windows, without WSL, if that is an option for you.

Best, Dan

fanzhongmeng, 2024-10-08:
Hi Daniel,
I have similar error as Andrew. I have checked my both WSL and conda installations, they are all in 64-bit.
Could it be that the openFAST installed in the conda environment via this command `conda install -y openfast` is the Windows version of openFAST?

Best regards
Fanzhong

dzalkind, 2024-10-08:
Hi Fanzhong,

Can you share your openfast version using `conda list | grep openfast`?

Best, Dan

fanzhongmeng, 2024-10-08:
Hi Daniel,
This is the openfast version:
`openfast                  3.5.3                h4226df5_0    conda-forge`
BR
Fanzhong

dzalkind, 2024-10-08:
That's definitely a linux version of `openfast`: https://anaconda.org/conda-forge/openfast/files?page=3

For a sanity check: can you execute that `openfast` executable?  Or same error?

I believe this is an issue with WSL and perhaps conda.  I recommend doing a web search of the issue with that context and try some solutions that have been found to work for other software.  One thing I'm seeing is that it can be an issue with the WSL version.  Do you know what version you're using?  Are you able to upgrade to WSL 2? Are you able to compile and run OpenFAST by itself within WSL?  Can you run other linux executables within WSL?

Sorry, the dev team doesn't use WSL (it's been banned by our IT), so this is a bit of a blind spot for us. Anything you find will be helpful to the community.

https://stackoverflow.com/questions/53353415/conda-commands-not-working-in-windows-subsystem-for-linux-wsl

fanzhongmeng, 2024-10-08:
Hi Daniel,
I can confirm that executing the openfast gives exactly the same error.
You are right. It could the issue of WSL. We both are using WSL 1.
I have not tried to compile and run openFAST within my WSL 1.
I can run other Linux executables within my WSL 1, for example, HAWC2 linux version.

BR
Fanzhong

Ajaru97, 2024-10-09:
Hi Both,

I have upgraded to WSL 2 and the examples are now able to run! Thank you for your help.

Just out of curiosity, what are the differences between WSL 1 and 2 that allowed them to run with WSL 2 but not with WSL1?

closed 2024-11-08 by dzalkind