NLRWindSystems/WEIS #319: RNA (Blade Mass Information)

hymdrj, 2024-11-13:
The RNA mass properties available in WEIS/RAFT includes mass, center of gravity, and rotational inertia. 
To calculate the mass and center of gravity of RNA, it’s essential to calculate the mass properties for each component, such as the blade and nacelle. 
Could you please let me know where can I find the code for these calculations? It seems like it should be in the RAFT_FOWT module, but I am currently unable to verify this.

gbarter, 2024-11-19:
Nacelle and RNA mass properties are inputs into RAFT, not outputs.  The properties are computed by WISDEM in its DrivetrainSE module.  You can find some details in the documentation [here](https://wisdem.readthedocs.io/en/master/wisdem/drivetrainse/components.html#nacelle-and-rna-mass-summary).

hymdrj, 2024-11-26:
Thank you for valuable comments. 

I occasionally see this warning message. What does it mean?

RuntimeWarning: C:\Users\Admin\miniforge3\envs\weis-env\Lib\site-packages\wisdem\commonse\utilization_dnvgl.py:322
The number of calls to function has reached maxfev = 50.

gbarter, 2024-11-26:
There are some internal root-finding or minimization routines within WISDEM and we limit the number of iterations in the backend scipy calls since we are content with engineering accuracy, not machine precision.  This is just an auto-generated warning that can likely be ignored (and likely to be disabled in the future).

hymdrj, 2024-11-27:
Thank you for your valuable comments. 

Also I have a question about NLBGS which solver for system constructing.

https://wisdem.readthedocs.io/en/master/examples/04_openmdao/tutorial.html#tutorial-2-the-sellar-problem
in figure 10.

Z: Global variables (dimensions of the turbine, tower, and substructure)
X: Local variables (variables limited to the turbine, tower, and substructure)
Y: Unknown variables

Discipline 1 and Discipline 2 are subsystems that make up the FOWT. Since the subsystems are interconnected, coupled variables such as Y emerge. The Gauss-Seidel method is employed to solve for the values of Y in the subsystems. Examples of Y include the platform's heave position or mooring tension.

Am I understanding this correctly?

Thank you for reading

hymdrj, 2024-11-28:
Hello @gbarter. 

Thank you very much for your kind response.   

Then, is there a reason to set the tolerance limit of NLBGS to 10^-8? Sometimes, during the modeling process, errors such as non-convergence occur due to too small a tolerance. Even if the tolerance is only at the level of 10^-5, it is considered to be very small. 

Please give your opinion.

gbarter, 2024-11-28:
At one point we noticed some differences with looser tolerances on nailing down behavior in Region II.5 in the power curve, but it has been a long time since we explored that and the code has evolved a bit since then.  You are welcome to adjust the convergence tolerance on your own.  This is just part of the space where model building and optimization is as much art as science.

closed 2025-02-14 by dzalkind