NLRWindSystems/WEIS #324: MPI logic is not working with RAFT (level 1) optimization

yonghoonlee, 2024-11-26:
## Description
The new MPI implementation utilizing the OpenMDAO's MPI object seems to work fine with OpenFAST examples, but it is not working with level 1 RAFT examples.

### Steps to reproduce issue
1. Install the current `develop` branch and move to `examples/15_RAFT_Studies` directory
2. Run `mpirun -np 4 python weis_driver_raft_opt.py`

### Current behavior
```
Using weis.aeroelasticse in rosco.toolbox...
Traceback (most recent call last):
  File "/Users/username/work/WEIS2/examples/15_RAFT_Studies/weis_driver_raft_opt.py", line 16, in <module>
Using weis.aeroelasticse in rosco.toolbox...
Traceback (most recent call last):
  File "/Users/username/work/WEIS2/examples/15_RAFT_Studies/weis_driver_raft_opt.py", line 16, in <module>
Using weis.aeroelasticse in rosco.toolbox...
Traceback (most recent call last):
  File "/Users/username/work/WEIS2/examples/15_RAFT_Studies/weis_driver_raft_opt.py", line 16, in <module>
Using weis.aeroelasticse in rosco.toolbox...
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
Traceback (most recent call last):
  File "/Users/username/work/WEIS2/examples/15_RAFT_Studies/weis_driver_raft_opt.py", line 16, in <module>
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/username/work/WEIS2/weis/glue_code/runWEIS.py", line 170, in run_weis
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/username/work/WEIS2/weis/glue_code/runWEIS.py", line 170, in run_weis
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, fname_modeling_options, fname_analysis_options)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/username/work/WEIS2/weis/glue_code/runWEIS.py", line 170, in run_weis
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/username/work/WEIS2/weis/glue_code/runWEIS.py", line 170, in run_weis
    wt_opt.setup(derivatives=False)
    wt_opt.setup(derivatives=False)
    wt_opt.setup(derivatives=False)
    wt_opt.setup(derivatives=False)
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/problem.py", line 1055, in setup
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/problem.py", line 1055, in setup
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/problem.py", line 1055, in setup
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/problem.py", line 1055, in setup
    model._setup(model_comm, self._metadata)
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/group.py", line 748, in _setup
    model._setup(model_comm, self._metadata)
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/group.py", line 748, in _setup
    model._setup(model_comm, self._metadata)
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/group.py", line 748, in _setup
    model._setup(model_comm, self._metadata)
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/group.py", line 748, in _setup
    self._setup_procs(self.pathname, comm, self._problem_meta)
    self._setup_procs(self.pathname, comm, self._problem_meta)
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/group.py", line 580, in _setup_procs
    self._setup_procs(self.pathname, comm, self._problem_meta)
    self._setup_procs(self.pathname, comm, self._problem_meta)
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/group.py", line 580, in _setup_procs
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/group.py", line 580, in _setup_procs
  File "/Users/username/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/group.py", line 580, in _setup_procs
    raise RuntimeError(msg)
RuntimeError: <model> <class Group>: num_par_fd = 4 but FD is not active.
    raise RuntimeError(msg)
    raise RuntimeError(msg)
RuntimeError: <model> <class Group>: num_par_fd = 4 but FD is not active.
    raise RuntimeError(msg)
RuntimeError: <model> <class Group>: num_par_fd = 4 but FD is not active.
RuntimeError: <model> <class Group>: num_par_fd = 4 but FD is not active.
```

### Expected behavior
Runs well with utilizing all cores in finite differencing

[... 270 lines cut ...]
typing-extensions         4.12.2               hd8ed1ab_0    conda-forge
typing_extensions         4.12.2             pyha770c72_0    conda-forge
tzdata                    2024b                hc8b5060_0    conda-forge
unicodedata2              15.1.0          py312h0bf5046_1    conda-forge
unixodbc                  2.3.12               h0e2417a_0    conda-forge
uri-template              1.3.0                    pypi_0    pypi
urllib3                   2.2.3              pyhd8ed1ab_0    conda-forge
wcwidth                   0.2.13                   pypi_0    pypi
webcolors                 24.11.1                  pypi_0    pypi
webencodings              0.5.1                    pypi_0    pypi
websocket-client          1.8.0                    pypi_0    pypi
weis                      1.4                      pypi_0    pypi
werkzeug                  3.0.6                    pypi_0    pypi
wheel                     0.45.1             pyhd8ed1ab_0    conda-forge
widgetsnbextension        4.0.13                   pypi_0    pypi
wisdem                    3.18.0          nompi_hedc07dd_101    conda-forge
xorg-libxau               1.0.11               hd74edd7_1    conda-forge
xorg-libxdmcp             1.1.5                hd74edd7_0    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
zeromq                    4.3.5                hc1bb282_7    conda-forge
zipp                      3.21.0             pyhd8ed1ab_0    conda-forge
zstandard                 0.23.0          py312h15fbf35_1    conda-forge
zstd                      1.5.6                hb46c0d2_0    conda-forge
```

yonghoonlee, 2024-11-26:
@dzalkind This is the issue that I discussed with you today. I will follow up with this issue once I figure out more details. Thank you.

dzalkind, 2024-11-26:
I think part of the issue here is that the `opt_flag` is not `True` here: https://github.com/WISDEM/WEIS/blob/08517c79b519579dbcb73f9963d70d04bf13befd/weis/glue_code/runWEIS.py#L165

If the optimization flag in the analysis options is set to True, this example works.

However, we should be able to detect the flags in the design variables.  Looking into that next.

dzalkind, 2024-11-26:
I think, based on the schema, this condition should always be true: https://github.com/WISDEM/WISDEM/blob/992076a14bf2f18defc1dd5be17e72c89e83431f/wisdem/glue_code/gc_LoadInputs.py#L650

@gbarter or @ptrbortolotti, has anything changed in this area or should we do that recursive search regardless?

gbarter, 2024-11-27:
> I think, based on the schema, this condition should always be true: https://github.com/WISDEM/WISDEM/blob/992076a14bf2f18defc1dd5be17e72c89e83431f/wisdem/glue_code/gc_LoadInputs.py#L650
> 
> @gbarter or @ptrbortolotti, has anything changed in this area or should we do that recursive search regardless?

I think that line predates the schema.  I agree that with the schema and enforcement of defaults, the check is unnecessary.

ptrbortolotti, 2024-11-27:
I also don't like it. I am not sure how that check made its way into the code, but essentially if any DV is set to be optimized, the opt solver flag is set to True regardless of the user input. do we want that? I like to have that master switch turning on and off optimizations in my analysis options yamls, especially because it helps switch between optimization and DOE. all in all, I'm in favor of changing those four line to
`self.analysis_options["opt_flag"] = self.analysis_options["driver"]["optimization"]["flag"]`, remove the if and more importantly the `else` and the `recursive_flag`

yonghoonlee, 2024-11-30:
@dzalkind I didn't pay attention to the schema before, but even if DV is flagged, it would be ideal to let users easily switch to do the optimization or not. Thank you for figuring this out.

yonghoonlee, 2024-11-30:
@dzalkind Another problem I wanted to bring up is that it behaves differently when MPI is used or not. I think the code should behave consistently whether or not the MPI is used.

dzalkind, 2024-12-02:
I agree with that.  We can address that in our upcoming MPI work.

What exactly is happening that's different?  Is it trying to optimize just because we're using MPI?

AthulKrishnaSundarrajan, 2024-12-03:
A different but related issue is when we need to run OpenFAST/DFSM simulations in parallel as part of the DOE studies. I'm adding a workaround I found for this issue, per @dzalkind's directions.

Running DOE studies with OpenFAST simulations in parallel hangs when generating the turbsim files. The workaround that runs just the openfast simulations in parallel for every DOE iteration is as follows:

1. Generate the turbsim files in serial, and copy and paste the 'wind' directory to the 'rank_0' directory in `OF_run_dir`.
2. Comment out this specific line in `openmdao_openfast.py`: [https://github.com/WISDEM/WEIS/blob/02cfd945dd49a617fbc385ffc4017b6d4f26b345/weis/aeroelasticse/openmdao_openfast.py#L374](https://github.com/WISDEM/WEIS/blob/02cfd945dd49a617fbc385ffc4017b6d4f26b345/weis/aeroelasticse/openmdao_openfast.py#L374)
3. In `runWEIS.py` : set n_FD and n_DV to be 1, update the code to be False in lines [line 89](https://github.com/WISDEM/WEIS/blob/02cfd945dd49a617fbc385ffc4017b6d4f26b345/weis/glue_code/runWEIS.py#L89) [line 139](https://github.com/WISDEM/WEIS/blob/02cfd945dd49a617fbc385ffc4017b6d4f26b345/weis/glue_code/runWEIS.py#L139) when running level3 simulations
4. Change [line 241](https://github.com/WISDEM/WEIS/blob/02cfd945dd49a617fbc385ffc4017b6d4f26b345/weis/glue_code/runWEIS.py#L241) from `(not opt_options['driver']['design_of_experiments']['flag'])` to `(opt_options['driver']['design_of_experiments']['flag'])` so that the other subprocessors stay up. 

It would be nice if this issue can also be fixed.

dzalkind, 2025-02-14:
Hi YongHoon,

We're actively testing this script now: https://github.com/WISDEM/WEIS/blob/b0738fe44072b7e58bbb87929baa687209912c94/.github/workflows/CI_WEIS.yml#L125

There have been some issues with the latest mpi4py, openmpi, openmdao, and petsc4py.  The installation instructions have been updated accordingly.

I hope this helps.

Best, Dan

yonghoonlee, 2025-02-14:
@dzalkind Thank you for your efforts working on this issue. I will update our code accordingly with the newest implementation of WEIS.
Bests, Yong Hoon

closed 2025-02-14 by yonghoonlee