NLRWindSystems/WEIS #325: Find the optimal turbine design after WEIS optimisation

ThomasPotentier, 2024-11-29:
## Description
I ran the example 02_run_openfast_cases, which i slightly modified to perform coupled optmisation on controller tuning and rotor diameter. It seems to be working as expected (with some limitations it appears),  as in the summary graphs I see metrics being updated.

1. I can't find the "optimal" solution in an OpenFAST format ? Or even the optimal turbine design (updated rotor diameter and tuned controller) in the WindIO format. Where is it stored ?
2. In the modelling_options file, if I turn the WISDEM -> RotorSE -> Flag: True there is an error message complaining about the Cp-Ct-Cq.txt table being too small compared to the study. If I switch the flaf to False, then there is no issue.
3. What is the purpose of 02_run_openfast_cases/IEA-15-240-RWT-outputs.yaml ? And how do I generate a new one a a new blade/turbine ?

I am uploading my input and output files to this issue (in txt format, as GitHub doesn't seem to accept the origianl yaml format):
Inputs:
* [analysis_options_controller_rotor_diam.txt](https://github.com/user-attachments/files/17957624/analysis_options_controller_rotor_diam.txt)
* [IEA-15-240-RWT.txt](https://github.com/user-attachments/files/17957625/IEA-15-240-RWT.txt)
* [modeling_options_controller_rotor_diam.txt](https://github.com/user-attachments/files/17957626/modeling_options_controller_rotor_diam.txt)

Summary output:
* [iea15mw-analysis.txt](https://github.com/user-attachments/files/17957645/iea15mw-analysis.txt)
* [iea15mw-modeling.txt](https://github.com/user-attachments/files/17957646/iea15mw-modeling.txt)
* [problem_vars.txt](https://github.com/user-attachments/files/17957647/problem_vars.txt)

I am using the latest WEIS version on WSL.

dzalkind, 2024-12-02:
There should be a directory called `openfast_runs` in your output directory where summary outputs are located.  That will contain the last set of openfast inputs that were run.  There _should_ be an `iea15mw.yaml` in that directory, but WEIS may not be writing the geometry output in this example because we bypass most of WISDEM in the 02 examples.

When the `Level3` -> `from_openfast` flag is true, WEIS will use the OpenFAST file specified in the modeling options and not change the geometry.  We primarily use them for running load cases or tuning controllers of _fixed_ turbine models.

I suggest you start with one of the examples in the 06_IEA-15-240-RWT directory, which will use WISDEM to generate OpenFAST inputs.  Then, you should be able to use WISDEM to change the rotor diameter.

The purpose of the `02_run_openfast_cases/IEA-15-240-RWT-outputs.yaml` is to provide OpenFAST with good initial conditions for simulations.  If you use WISDEM, you should not need this.  @ptrbortolotti, are there any scripts available to generate one of these output yamls?

To use OpenFAST to minimize the thrust, you will need to [create an output in this component](https://github.com/WISDEM/WEIS/blob/08517c79b519579dbcb73f9963d70d04bf13befd/weis/aeroelasticse/openmdao_openfast.py#L422).  You should be able to [use the summary stats around here](https://github.com/WISDEM/WEIS/blob/08517c79b519579dbcb73f9963d70d04bf13befd/weis/aeroelasticse/openmdao_openfast.py#L2159) to get the maximum thrust from OpenFAST runs.

I hope this helps.  Thank you for the detailed questions and inputs.  It's helpful!  Please let us know if you need any further assistance.

Best, Dan

ThomasPotentier, 2024-12-04:
Dear Dan,

Thank you for your detailed and precise explanation. 

I am currently busy on another topic, as soon as it is done, I will go back to WEIS and apply your advice.

Best,

Thomas

dzalkind, 2025-05-14:
Please let us know if you have additional questions or would like support in setting up new workflows.

closed 2025-05-14 by dzalkind