NLRWindSystems/WEIS #326: Error when running weis_driver.py from 03_NREL5MW_OC3_spar

MalikSpahic, 2024-12-06:
Hello,

I am making this ticket, because after following the installation procedure given in the github for WEIS on a virtual machine and linux environment, I get an error when I run the script 'weis_driver.py' from the folder '03_NREL5MW_OC3_spar'. However, I am not witnessing any error when I run 'weis_freq_driver.py'.

Please, find below the description of the bug:

## Description
_Describe the bug here_

 <model> <class WindPark>: Attempted to connect from 'aeroelastic.hub_Fxyz' to 'drivese_post.F_hub', but 'drivese_post.F_hub' doesn't exist. Perhaps you meant to connect to one of the following inputs: ['drivese_post.hub_I', 'drivese_post.F_mb2', 'drivese_post.F_mb1'].
   <model> <class WindPark>: Attempted to connect from 'aeroelastic.hub_Mxyz' to 'drivese_post.M_hub', but 'drivese_post.M_hub' doesn't exist. Perhaps you meant to connect to one of the following inputs: ['drivese_post.hub_I', 'drivese_post.M_mb2', 'drivese_post.M_mb1'].

Could you help me fixing it ?

Kind regards,

Malik

dzalkind, 2024-12-06:
Hi Malik,

Do you mind posting your conda environment?  `conda list` 

I suspect the issue is that an older version of wisdem was installed.  Please try `conda install wisdem=3.18`.

I will post a fix to the environment.yml file so this issue does not occur in the future.  Thank you for posting!

Best, Dan

MalikSpahic, 2024-12-10:
Hey Dan,

Thank you for your fast answer ! Sure here is my conda environment :

# packages in environment at /home/ubuntu/anaconda3/envs/envWeis:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                  2_kmp_llvm    conda-forge
alsa-lib                  1.2.12               h4ab18f5_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
anyio                     4.6.2.post1              pypi_0    pypi
argon2-cffi               23.1.0                   pypi_0    pypi
argon2-cffi-bindings      21.2.0                   pypi_0    pypi
arrow                     1.3.0                    pypi_0    pypi
asttokens                 2.4.1                    pypi_0    pypi
async-lru                 2.0.4                    pypi_0    pypi
attrs                     24.2.0             pyh71513ae_0    conda-forge
babel                     2.16.0                   pypi_0    pypi
beautifulsoup4            4.12.3                   pypi_0    pypi
bleach                    6.1.0                    pypi_0    pypi
blinker                   1.8.2              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hb9d3cd8_2    conda-forge
brotli-bin                1.1.0                hb9d3cd8_2    conda-forge
brotli-python             1.1.0           py312h2ec8cdc_2    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.34.2               heb4867d_0    conda-forge
ca-certificates           2024.8.30            hbcca054_0    conda-forge
cairo                     1.18.0               hebfffa5_3    conda-forge
certifi                   2024.8.30          pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py312h06ac9bb_0    conda-forge
charset-normalizer        3.4.0              pyhd8ed1ab_0    conda-forge
click                     8.1.7           unix_pyh707e725_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
comm                      0.2.2                    pypi_0    pypi
contourpy                 1.3.0           py312h68727a3_2    conda-forge
control                   0.10.1             pyhd8ed1ab_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
cyrus-sasl                2.1.27               h54b06d7_7    conda-forge
dash                      2.18.1             pyhd8ed1ab_0    conda-forge
dash-bootstrap-components 1.6.0                    pypi_0    pypi
dash-core-components      2.0.0                    pypi_0    pypi
dash-html-components      2.0.0                    pypi_0    pypi
dash-mantine-components   0.14.6                   pypi_0    pypi
dash-table                5.0.0                    pypi_0    pypi
dbus                      1.13.6               h5008d03_3    conda-forge
dearpygui                 2.0.0                    pypi_0    pypi
debugpy                   1.8.7                    pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
defusedxml                0.7.1                    pypi_0    pypi
dill                      0.3.9              pyhd8ed1ab_0    conda-forge
docopt                    0.6.2                    pypi_0    pypi
double-conversion         3.3.0                h59595ed_0    conda-forge
et_xmlfile                1.1.0              pyhd8ed1ab_0    conda-forge
exceptiongroup            1.2.2              pyhd8ed1ab_0    conda-forge
executing                 2.1.0                    pypi_0    pypi
expat                     2.6.3                h5888daf_0    conda-forge
fastjsonschema            2.20.0                   pypi_0    pypi
fatpack                   0.7.7              pyhd8ed1ab_0    conda-forge
fftw                      3.3.10          mpi_openmpi_h99e62ba_10    conda-forge
flask                     3.0.3              pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_3    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.54.1          py312h66e93f0_0    conda-forge
fqdn                      1.5.1                    pypi_0    pypi
[... 282 lines cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.5                h3b0a872_6    conda-forge
zipp                      3.20.2             pyhd8ed1ab_0    conda-forge
zlib                      1.3.1                hb9d3cd8_2    conda-forge
zstandard                 0.23.0          py312hef9b889_1    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge

As you guessed, it seems that the wisdem version that I was using was old (wisdem v. 3.17). But now, I got a new error for both python scripts "weis_driver.py" and "weis_freq_driver.py" : 

Using weis.aeroelasticse in rosco.toolbox...
Traceback (most recent call last):
[... cut ...] req_driver.py", line 5, in <module>
    from weis.glue_code.runWEIS     import run_weis
  File "/home/ubuntu/WEIS/weis/glue_code/runWEIS.py", line 4, in <module>
    from weis.glue_code.gc_LoadInputs     import WindTurbineOntologyPythonWEIS
  File "/home/ubuntu/WEIS/weis/glue_code/gc_LoadInputs.py", line 10, in <module>
    from wisdem.commonse.mpi_tools              import MPI
ModuleNotFoundError: No module named 'wisdem.commonse.mpi_tools'

Could you help me to solve it ?

Kind regards,

Malik

dzalkind, 2024-12-11:
Hi Malik,

Please pull in the latest `main` branch, where we have updated these `wisdem` imports.

Best, Dan

MalikSpahic, 2025-01-13:
Hi Dan,

Thank you for your answer and sorry for the late reply (I was on vacations soon after your message). I downloaded the 1.4 version and the error is not present for this example anymore :)

Thank you for fixing it that fast !

Kind regards,

Malik

closed 2025-02-14 by dzalkind