NLRWindSystems/WEIS #346: Python scripts errors in the folder '01_aeroelasticse'

MalikSpahic, 2025-01-21:
Hi there,

I tried to run the examples again with the latest version of Weis (v1.4). Some example python scripts from the folder '01_aeroelasticse' are not working: the scripts 'run_nodalOutputs.py' and 'run_general.py'. For the error generated by 'run_general.py', the issue can be fixed by creating a folder with the right version of the 'libdiscon.so' file. At the following location: ./WEIS/local/lib/libdiscon.so. Would it be possible to add it in a new/modified version of the WEIS folder please ?

Please find below the description of the errors:

## Description
Error messages when running 'run_nodalOutputs.py' and 'run_general.py' from the folder '01_aeroelasticse'

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. Activate your Weis conda environment
2. Go to the folder location './WEIS/examples/01_aeroelasticse'
3. Run the scripts 'run_nodalOutputs.py' and 'run_general.py'

### Current behavior

**For 'run_nodalOutput.py' the error message is:**

Using weis.aeroelasticse in rosco.toolbox...
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "/home/ubuntu/anaconda3/envs/eWeis1p4/lib/python3.12/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/anaconda3/envs/eWeis1p4/lib/python3.12/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
           ^^^^^^^^^^^^^^^^
  File "/home/ubuntu/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 510, in evaluate_multi
    return evaluate(indict)
           ^^^^^^^^^^^^^^^^
  File "/home/ubuntu/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 505, in evaluate
    return fast.execute()
           ^^^^^^^^^^^^^^
  File "/home/ubuntu/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 167, in execute
    reader.execute()
  File "/home/ubuntu/WEIS/weis/aeroelasticse/FAST_reader.py", line 2524, in execute
    self.read_AeroDyn15()
  File "/home/ubuntu/WEIS/weis/aeroelasticse/FAST_reader.py", line 848, in read_AeroDyn15
    self.fst_vt['AeroDyn15']['MaxIter']               = int(f.readline().split()[0])
                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '======'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/ubuntu/WEIS/examples/01_aeroelasticse/run_nodalOutputs.py", line 86, in <module>
    fastBatch.run_multi(n_cores)
  File "/home/ubuntu/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 411, in run_multi
    output = pool.map(evaluate_multi, case_data_all)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/anaconda3/envs/eWeis1p4/lib/python3.12/multiprocessing/pool.py", line 367, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/anaconda3/envs/eWeis1p4/lib/python3.12/multiprocessing/pool.py", line 774, in get
    raise self._value
ValueError: invalid literal for int() with base 10: '======'

----------------------------------------------------------------------------------------------------------------------------------------------------------------

----------------------------------------------------------------------------------------------------------------------------------------------------------------

**For 'run_general.py' the error message is:**

Using weis.aeroelasticse in rosco.toolbox...
Loading rotor performace data from text file: /home/ubuntu/WEIS/examples/01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-RWT/Cp_Ct_Cq.IEA15MW.txt
Writing rotor performance text file: steady_state/iea15mw/iea15mw_0_Cp_Ct_Cq.txt
Writing new controller parameter file parameter file: steady_state/iea15mw/iea15mw_0_DISCON.IN.

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2024 National Renewable Energy Laboratory
 Copyright (C) 2024 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST--128-NOTFOUND
 Compile Info:
  - Compiler: GCC version 13.3.0
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: Yes, number of threads: 24/24
  - Date: Nov  2 2024
  - Time: 12:54:10
 Execution Info:
  - Date: 01/21/2025
  - Time: 10:08:18+0100

 OpenFAST input file heading:
     Generated with AeroElasticSE FAST driver

 Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
 Running AeroDyn.
 AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
 Skipping nodal outputs.
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 1, blade 1)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 2, blade 1)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 1, blade 2)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 2, blade 2)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 1, blade 3)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 2, blade 3)
 Running InflowWind.
 Running HydroDyn.
  Generating incident wave kinematics and current time history.
 Running SubDyn.
 !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
 Warning: the SubDyn input file is not at the latest format!
          Visit: https://openfast.readthedocs.io/en/dev/source/user/api_change.html
 > Issue: Member table contains 6 columns instead of 7,  using default member directional cosines
 ID (-1) for all members. The directional cosines will be computed based on the member nodes for
 all members.
 !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
 !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
 Warning: the SubDyn input file is not at the latest format!
          Visit: https://openfast.readthedocs.io/en/dev/source/user/api_change.html
 > Issue: OutCBModes and OutFEMModes are not present in input file towards the output section
 !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
    Fixed bottom case detected
    Performing Craig-Bampton reduction 60 DOFs -> 0 modes + 6 DOFs
    Using static improvement method for gravity and ext. loads
    Calculating Full System Modes for output files
    Exporting Summary file
 Running ServoDyn.
 Running ServoDyn Interface for Bladed Controllers (using GNU Fortran for Linux).
Traceback (most recent call last):
  File "/home/ubuntu/WEIS/examples/01_aeroelasticse/run_general.py", line 80, in <module>
    fastBatch.run_serial()
  File "/home/ubuntu/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 386, in run_serial
    _name, _ss, _et, _dl, _dam, _ct = evaluate(c)
                                      ^^^^^^^^^^^
  File "/home/ubuntu/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 505, in evaluate
    return fast.execute()
           ^^^^^^^^^^^^^^
  File "/home/ubuntu/WEIS/weis/aeroelasticse/runFAST_pywrapper.py", line 198, in execute
    openfastlib.fast_run()
  File "/home/ubuntu/WEIS/weis/aeroelasticse/openfast_library.py", line 240, in fast_run
    self.fast_init()
  File "/home/ubuntu/WEIS/weis/aeroelasticse/openfast_library.py", line 160, in fast_init
    raise RuntimeError(f"Error {_error_status.value}: {_error_message.value}")
RuntimeError: Error 4: b'FAST_InitializeAll:HydroDyn_Init:Waves_Init:VariousWaves_Init: The random number generator in use differs from the original code provided by NREL. This pRNG uses 8 seeds instead of the 2 in the HydroDyn input file.\nFAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library /home/ubuntu/WEIS/local/lib/libdiscon.so could not be loaded. Check that the file exists in the specified location and that it is compiled for 64-bit applications.'

Kind regards,

Malik

gbarter, 2025-01-22:
I think the scripts work in the `develop` and some feature branches, but those have some larger changes queued up as well that Malik might not want to swallow quite yet.

dzalkind, 2025-02-14:
Hi Malik,

These examples are now working (and tested) in the latest release.

closed 2025-02-14 by dzalkind