NLRWindSystems/WEIS #353: Duplicate generate_wind_file function

dzalkind, 2025-02-18:
## Description
On the `main` branch there are two versions of the same function:

- https://github.com/WISDEM/WEIS/blob/b0738fe44072b7e58bbb87929baa687209912c94/weis/aeroelasticse/turbsim_util.py#L223
- https://github.com/WISDEM/WEIS/blob/b0738fe44072b7e58bbb87929baa687209912c94/weis/aeroelasticse/utils.py#L516

On the `develop` branch, there is only 1 version:
- https://github.com/WISDEM/WEIS/blob/fa3c60949b01b5b95dea6eb03e9ebc39a8918bf2/weis/aeroelasticse/utils.py#L516

I'm guessing that the `develop` branch has it right.  Note that the `new_pcrunch` branch is like the `main` branch.  

I don't think this is breaking anything, but if you're developing in that function, heads up.  I'll clean this up.

If anyone has ideas about how we got here and strong opinions about where it should live, please let me know.

ptrbortolotti, 2025-02-18:
`dev` is right, `turbsim_util` is now in openfast_io [link](https://github.com/OpenFAST/openfast/blob/main/openfast_io/openfast_io/turbsim_util.py) without `generate_wind_files`

ptrbortolotti, 2025-02-18:
I don't see issues with `new_pcrunch`...

ptrbortolotti, 2025-03-25:
I believe this can be closed as completed @dzalkind

closed 2025-03-26 by dzalkind