NLRWindSystems/WEIS #354: Conda installation issue - suspecting dependency issue

yonghoonlee, 2025-02-19:
## Description

The newest version (both v1.5 on main branch and the latest develop branch) cannot be installed on Linux machine, due to conda error.

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. Follow the installation instructions:
```
git clone https://github.com/WISDEM/WEIS
cd WEIS
git checkout develop
conda env create --name weis-env -f environment.yml
```

### Current behavior
```
Channels:
 - conda-forge
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages:

Preparing transaction: failed

# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/exception_handler.py", line 18, in __call__
        return func(*args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/cli/main.py", line 61, in main_subshell
        exit_code = do_call(args, parser)
                    ^^^^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/cli/conda_argparse.py", line 205, in do_call
        result = getattr(module, func_name)(args, parser)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/notices/core.py", line 132, in wrapper
        return func(*args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/cli/main_env_create.py", line 178, in execute
        result[installer_type] = installer.install(
                                 ^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/env/installers/conda.py", line 67, in install
        unlink_link_transaction.execute()
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/core/link.py", line 342, in execute
        self.verify()
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/common/io.py", line 87, in decorated
        return f(*args, **kwds)
               ^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/core/link.py", line 289, in verify
        self.prepare()
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/core/link.py", line 273, in prepare
        grps = self._prepare(
               ^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/core/link.py", line 403, in _prepare
        packages_info_to_link = tuple(
                                ^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/core/link.py", line 404, in <genexpr>
        read_package_info(prec, pcrec)
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/gateways/disk/read.py", line 93, in read_package_info
        paths_data = read_paths_json(epd)
                     ^^^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/site-packages/conda/gateways/disk/read.py", line 169, in read_paths_json
        data = json.load(paths_json)
               ^^^^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/json/__init__.py", line 293, in load
        return loads(fp.read(),
               ^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/json/__init__.py", line 346, in loads
        return _default_decoder.decode(s)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/json/decoder.py", line 337, in decode
        obj, end = self.raw_decode(s, idx=_w(s, 0).end())
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/home/stamang1/miniforge3/lib/python3.12/json/decoder.py", line 355, in raw_decode
        raise JSONDecodeError("Expecting value", s, err.value) from None
    json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

`$ /home/stamang1/miniforge3/bin/conda env create --name weis-env -f environment.yml`

  environment variables:
                 CIO_TEST=<not set>
                CONDA_EXE=/home/stamang1/miniforge3/bin/conda
         CONDA_PYTHON_EXE=/home/stamang1/miniforge3/bin/python
               CONDA_ROOT=/home/stamang1/miniforge3
              CONDA_SHLVL=0
           CURL_CA_BUNDLE=<not set>
               LD_PRELOAD=<not set>
                     PATH=/home/stamang1/miniforge3/condabin:/usr/local/sbin:/usr/local/bin:/usr
                          /sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
       REQUESTS_CA_BUNDLE=<not set>
            SSL_CERT_FILE=<not set>
             SUDO_COMMAND=/usr/bin/su stamang1
                 SUDO_GID=1000
                 SUDO_UID=1000
                SUDO_USER=lee

     active environment : None
            shell level : 0
       user config file : /home/stamang1/.condarc
 populated config files : /home/stamang1/miniforge3/.condarc
                          /home/stamang1/.condarc
          conda version : 25.1.1
    conda-build version : not installed
         python version : 3.12.5.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=cascadelake
                          __conda=25.1.1=0
                          __glibc=2.35=0
                          __linux=6.8.0=0
                          __unix=0=0
       base environment : /home/stamang1/miniforge3  (writable)
      conda av data dir : /home/stamang1/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/stamang1/miniforge3/pkgs
                          /home/stamang1/.conda/pkgs
       envs directories : /home/stamang1/miniforge3/envs
                          /home/stamang1/.conda/envs
               platform : linux-64
             user-agent : conda/25.1.1 requests/2.32.3 CPython/3.12.5 Linux/6.8.0-52-generic ubuntu/22.04.5 glibc/2.35 solver/libmamba conda-libmamba-solver/25.1.1 libmambapy/2.0.5
                UID:GID : 1004:1004
             netrc file : None
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.
If you suspect this error is being caused by a malfunctioning plugin,
consider using the --no-plugins option to turn off plugins.

Example: conda --no-plugins install <package>

Alternatively, you can set the CONDA_NO_PLUGINS environment variable on
the command line to run the command without plugins enabled.

Example: CONDA_NO_PLUGINS=true conda install <package>

If submitted, this report will be used by core maintainers to improve
future releases of conda.
Would you like conda to send this report to the core maintainers? [y/N]: n

No report sent. To permanently opt-out, use

    $ conda config --set report_errors false
```

### Expected behavior
Installation success.

### Code versions
 - conda=25.1.1

yonghoonlee, 2025-02-19:
This issue was originally raised by @amsunilsunil and I was also able to reproduce the same error.

dzalkind, 2025-02-19:
This seems like an issue with your local conda set up Can you try the suggestion here? https://github.com/conda/conda/issues/13026#issuecomment-1694138646

yonghoonlee, 2025-03-05:
The issue was resolved by following the direction given by @dzalkind. Thank you for helping us. I will now close the issue.

closed 2025-03-05 by yonghoonlee