NLRWindSystems/WEIS #363: RAFT example not working - OpenMDAO variable connections broken

yonghoonlee, 2025-03-14:
## Description
Failure in running RAFT optimization example.
It seems that the OpenMDAO variable connections are broken.

### Steps to reproduce issue
Installed the current (as of March 14, 2025) version of WEIS develop branch with commit hash: `df651c32c2cb6cc9f198a41d5b9fc4341df8f9e8` and ran:
```
cd examples/15_RAFT_Studies
python weis_driver_raft_opt.py
```

### Current behavior
```
Using weis.aeroelasticse in rosco.toolbox...
WARNING: AggregatedJoints hasn't included rectangular yet
WARNING: AggregatedJoints hasn't included rectangular yet
WARNING: AggregatedJoints hasn't included rectangular yet
Traceback (most recent call last):
  File "/Users/yhlee/work/WEIS/examples/15_RAFT_Studies/weis_driver_raft_opt.py", line 11, in <module>
    wt_opt, modeling_options, opt_options = weis_main(fname_wt_input,
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yhlee/work/WEIS/weis/main.py", line 48, in weis_main
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, 
                                            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yhlee/work/WEIS/weis/glue_code/runWEIS.py", line 117, in run_weis
    wt_opt.setup()
  File "/Users/yhlee/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/problem.py", line 1064, in setup
    self._check_collected_errors()
  File "/Users/yhlee/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/problem.py", line 636, in _check_collected_errors
    raise RuntimeError('\n'.join(final_msg))
RuntimeError: 
Collected errors for problem 'weis_driver_raft_opt':
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'sse_tune.Rtip', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'sse_tune.tune_rosco.R', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'raft.blade_Rtip', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
```

### Expected behavior

### Code versions
```
# Name                    Version                   Build  Channel
alabaster                 1.0.0              pyhd8ed1ab_1    conda-forge
ampl-asl                  1.0.0                h286801f_2    conda-forge
anyio                     4.8.0                    pypi_0    pypi
appnope                   0.1.4                    pypi_0    pypi
argon2-cffi               23.1.0                   pypi_0    pypi
argon2-cffi-bindings      21.2.0                   pypi_0    pypi
arrow                     1.3.0                    pypi_0    pypi
asttokens                 3.0.0                    pypi_0    pypi
async-lru                 2.0.4                    pypi_0    pypi
attrs                     25.3.0             pyh71513ae_0    conda-forge
babel                     2.17.0             pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.13.3                   pypi_0    pypi
bleach                    6.2.0                    pypi_0    pypi
blinker                   1.9.0              pyhff2d567_0    conda-forge
brotli                    1.1.0                hd74edd7_2    conda-forge
brotli-bin                1.1.0                hd74edd7_2    conda-forge
brotli-python             1.1.0           py312hde4cb15_2    conda-forge
bzip2                     1.0.8                h99b78c6_7    conda-forge
c-ares                    1.34.4               h5505292_0    conda-forge
ca-certificates           2025.1.31            hf0a4a13_0    conda-forge
certifi                   2025.1.31          pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py312h0fad829_0    conda-forge
charset-normalizer        3.4.1              pyhd8ed1ab_0    conda-forge
click                     8.1.8              pyh707e725_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_1    conda-forge
comm                      0.2.2                    pypi_0    pypi
contourpy                 1.3.1           py312hb23fbb9_0    conda-forge
control                   0.10.1             pyhd8ed1ab_1    conda-forge
[... 255 lines cut ...]
treon                     0.1.4                    pypi_0    pypi
types-python-dateutil     2.9.0.20241206           pypi_0    pypi
typing_extensions         4.12.2             pyha770c72_1    conda-forge
tzdata                    2025a                h78e105d_0    conda-forge
unicodedata2              16.0.0          py312hea69d52_0    conda-forge
uri-template              1.3.0                    pypi_0    pypi
urllib3                   2.3.0              pyhd8ed1ab_0    conda-forge
vtk                       9.3.1                    pypi_0    pypi
wcwidth                   0.2.13                   pypi_0    pypi
webcolors                 24.11.1                  pypi_0    pypi
webencodings              0.5.1                    pypi_0    pypi
websocket-client          1.8.0                    pypi_0    pypi
weis                      1.4                      pypi_0    pypi
werkzeug                  3.1.3              pyhd8ed1ab_1    conda-forge
wheel                     0.45.1             pyhd8ed1ab_1    conda-forge
widgetsnbextension        4.0.13                   pypi_0    pypi
wisdem                    3.20.0          nompi_hedc07dd_100    conda-forge
xorg-libxau               1.0.12               h5505292_0    conda-forge
xorg-libxdmcp             1.1.5                hd74edd7_0    conda-forge
yaml                      0.2.5                h3422bc3_2    conda-forge
zeromq                    4.3.5                hc1bb282_7    conda-forge
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zstandard                 0.23.0          py312hea69d52_1    conda-forge
zstd                      1.5.7                h6491c7d_1    conda-forge
```

yonghoonlee, 2025-03-14:
Besides the issue raised above, the following change needs to be made:

https://github.com/WISDEM/WEIS/blob/df651c32c2cb6cc9f198a41d5b9fc4341df8f9e8/examples/15_RAFT_Studies/modeling_options_OpenFAST.yaml#L81

```
-    tuning_yaml: ../01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-RWT-UMaineSemi/IEA15MW-UMaineSemi.yaml
+    tuning_yaml: ../01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_ROSCO.yaml
```

yonghoonlee, 2025-03-15:
The issue is not only appearing with the RAFT, but also appearing with the OpenFAST cases.
Please refer to another case I have tried, given below:

```
cd examples/06_IEA-15-240-RWT/
python weis_driver_umaine_semi.py
```

```
Using weis.aeroelasticse in rosco.toolbox...
Found existing potential model: ../01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-RWT-UMaineSemi/HydroData/IEA-15-240-RWT-UMaineSemi
    - Trying to use this instead of running PyHAMS.
Traceback (most recent call last):
  File "/Users/yhlee/work/WEIS/examples/06_IEA-15-240-RWT/weis_driver_umaine_semi.py", line 10, in <module>
    wt_opt, modeling_options, opt_options = weis_main(fname_wt_input,
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yhlee/work/WEIS/weis/main.py", line 48, in weis_main
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, 
                                            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yhlee/work/WEIS/weis/glue_code/runWEIS.py", line 121, in run_weis
    wt_opt.setup(derivatives=False)
  File "/Users/yhlee/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/problem.py", line 1064, in setup
    self._check_collected_errors()
  File "/Users/yhlee/miniforge3/envs/weis-env/lib/python3.12/site-packages/openmdao/core/problem.py", line 636, in _check_collected_errors
    raise RuntimeError('\n'.join(final_msg))
RuntimeError: 
Collected errors for problem 'weis_driver_umaine_semi':
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'sse_tune.Rtip', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'sse_tune.tune_rosco.R', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'aeroelastic.Rtip', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'tcons_post.Rtip', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
```

yonghoonlee, 2025-03-15:
I also tried with the main branch (v1.5.1 after pCrunch pinning commit, with commit hash `0da256fe7b02de10cf0726830b3f1da26c087c8f`) and this version has the same issue.

```
cd examples/06_IEA-15-240-RWT
python weis_driver_umaine_semi.py
```

```
Using weis.aeroelasticse in rosco.toolbox...
Found existing potential model: ../01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-RWT-UMaineSemi/HydroData/IEA-15-240-RWT-UMaineSemi
    - Trying to use this instead of running PyHAMS.
Traceback (most recent call last):
  File "/Users/yhlee/work/WEIS-0da256f/examples/06_IEA-15-240-RWT/weis_driver_umaine_semi.py", line 10, in <module>
    wt_opt, modeling_options, opt_options = weis_main(fname_wt_input,
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yhlee/work/WEIS-0da256f/weis/main.py", line 48, in weis_main
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, 
                                            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yhlee/work/WEIS-0da256f/weis/glue_code/runWEIS.py", line 121, in run_weis
    wt_opt.setup(derivatives=False)
  File "/Users/yhlee/miniforge3/envs/weis-0da256f/lib/python3.12/site-packages/openmdao/core/problem.py", line 1064, in setup
    self._check_collected_errors()
  File "/Users/yhlee/miniforge3/envs/weis-0da256f/lib/python3.12/site-packages/openmdao/core/problem.py", line 636, in _check_collected_errors
    raise RuntimeError('\n'.join(final_msg))
RuntimeError: 
Collected errors for problem 'weis_driver_umaine_semi':
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'sse_tune.Rtip', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'sse_tune.tune_rosco.R', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'aeroelastic.Rtip', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
   <model> <class WindPark>: Attempted to connect from 'blade.high_level_blade_props.rotor_radius' to 'tcons_post.Rtip', but 'blade.high_level_blade_props.rotor_radius' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['blade.high_level_blade_props.rotor_solidity', 'blade.high_level_blade_props.rotor_diameter', 'blade.high_level_blade_props.r_blade'].
```

gbarter, 2025-03-15:
@yonghoonlee - Can you please double check that your WISDEM and WEIS files are synced with the latest versions?

yonghoonlee, 2025-03-17:
@gbarter I created new environments every time I install them, so I think the WISDEM is installed as the environment.yml specifies. The WISDEM version is 3.20.0, which seems to be the latest release version, and installed through conda-forge repository.
Also, the WEIS code was cloned directly from the https://github.com/WISDEM/WEIS.git, so the WEIS version is the latest one. I tried both the `develop` and `main` branches. They behave the same way.

gbarter, 2025-03-19:
Sorry for the delay.  I can confirm I can recreate the error.

@ptrbortolotti - Most of the broken connections I think arose when you resolved the ambiguity around rotor diameter.  There are 4-5 places in the WEIS glue code that need to be fixed (not sure why the tests didn't pick this up).  All but one of them should clearly be changed from `rotor_radius` (which no longer exists) to `Rtip`.  I cannot tell about this line though:
https://github.com/WISDEM/WEIS/blob/main/weis/glue_code/glue_code.py#L161

What do you think?

gbarter, 2025-03-19:
@ptrbortolotti - Sorry, it looks like you already resolved this issue at least in your `precomp2beamdyn` branch.  I can either push these specific changes in a new branch or you can merge your branch into develop.  I will see if I can recreate the other issue that is holding up your branch.

gbarter, 2025-03-19:
@yonghoonlee - Pietro pointed out that we just needed to better lock with WISDEM version with the various WEIS branches.  I have updated the `environment.yml` in the develop and main branches to WISDEM v3.19.  All you need to do is downgrade via `conda install wisdem==3.19.0`

yonghoonlee, 2025-03-19:
@gbarter Thank you for resolving the issue. I will downgrade and test it. I believe it should work, so I will close the issue. Thank you for your help.

closed 2025-03-19 by yonghoonlee