NLRWindSystems/WEIS #364: How do you not lock root of the chord in blade opt

TheMercer, 2025-03-17:
`            chord:
                flag: True             # Flag to optimize the chord
                n_opt: 16               # Number of control points along blade span
                max_decrease: 0.3      # Minimum multiplicative gain on existing chord at the n_opt locations
                max_increase: 3.       # Maximum multiplicative gain on existing chord at the n_opt locations
                index_start: 2         # Lock the first two DVs from blade root
                index_end: 16           # All DVs close to blade tip are active
`
index_start: 2         # Lock the first two DVs from blade root 
^ this line, how do I stop it locking do I just remove it ?

gbarter, 2025-03-17:
You can either remove it, or for better transparency, set to zero.

TheMercer, 2025-03-17:
Cheers will try that

TheMercer, 2025-03-17:
index_start: 0         # Lock the first two DVs from blade root

Seems to keep keep the first value of the chord the same

gbarter, 2025-03-17:
The root diameter must interface with the hub and is usually a fixed target.  You generally don't want the optimizer to play with that value.  WISDEM does not capture all of the structural requirements for sizing the hub interface with the root chord and thickness, so we suggest leaving that fixed.

That is also why the initial `index_start` was set to 2 as it takes a bit of span length to morph from the blade root cylinder to something resembling an airfoil shape.

closed 2025-03-17 by TheMercer