NLRWindSystems/WEIS #365: Precomp2Beamdyn

TheMercer, 2025-03-18:
## Description
The beamdyn output is not saved in WindIO output

### Steps to reproduce issue
1.run example 6, Beamdyn files are generated, however output windIO file still has precomp
2. …
3. …

### Current behavior
… windIO file still has precompvalues

### Expected behavior
… windIO file still has Beamdyn values

TheMercer, 2025-03-18:
I have also noted, I only ran one custom setup on this PR. The edge stiffness was much lower then flap stiffness in general along the blade. Causing issues with the blade. This however could be the setup I had (Just thought I would note it)

ptrbortolotti, 2025-03-18:
You raise an excellent point @TheMercer , I will move the rototranslations of K and I from WEIS to WISDEM so that the properties can be printed out to the output `yaml` file in a simple WISDEM call and can be passed to WEIS. This will require a new tag of WISDEM, but it should be a quick upgrade. I'll keep this issue open

TheMercer, 2025-03-18:
Additional, in the comparison of torsional of precomp on other software/sets of code. 

What's the comparison of Edge, Flap, Axial at the same time ?

ptrbortolotti, 2025-03-18:
You can test things right now by following these steps:
- Remove `WISDEM` from your WEIS environment: `conda remove wisdom --force`
- In your favorite directory, clone `WISDEM`: `git clone git@github.com:WISDEM/WISDEM.git`
- Switch to the `windio_KI` branch: `git checkout wind_KI`
- Install `WISDEM` in your `WEIS` environment: `pip install --no-deps -e . -v`
- Use the latest commit on `WEIS` branch `precomp2beamdyn`: `git fetch origin` and `git checkout precomp2beamdyn`

ptrbortolotti, 2025-03-18:
The comparison for the IEA15 is this one (I left out off-diagonal terms where ANBA or both solvers returned zeros)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

TheMercer, 2025-03-18:
I will tried the wisdem update tomorrow morning thank you,

Yeah, the match between them looks really good. 

I will have a look into mine, after updating wisdem maybe I copied it wrong. If it's still persists. I will double check my setup. Maybe open an new seperate ticket.

Thank you for the support on this.

closed 2025-03-19 by TheMercer