NLRWindSystems/WEIS #366: Installing WEIS on native Windows

mcollu, 2025-03-20:
# Description of feature
Installing WEIS on native Windows, rather than using WSL.
Although WSL seems to work well, based on my limited experience, I am collaborating with some companies where the the internal IT policy could prevent the from using WSL.
I would like to know if there is any update on this front.

# Potential solution
Not sure, not my field.

ptrbortolotti, 2025-03-20:
hi @mcollu , I believe my NREL colleagues have already overcome this limitation, and WEIS should now compile successfully on Windows. I recently updated the README reflecting this change. Are you having issues?

ptrbortolotti, 2025-03-20:
(The change of the readme hasn't made its way to the master branch just yet...)

mcollu, 2025-03-20:
Hi @ptrbortolotti , to be honest I was trying to find it and not able to, and based on the communications I was able to find, I thought it was not released yet? Where can I find the instructions on how to install natively on windows?

gbarter, 2025-03-20:
I suggest following Steps 2 & 3 [here](https://github.com/WISDEM/WEIS/tree/develop).  On Windows, in Step 3, I would just install `pyoptsparse` and don't install `mpi4py` or `petsc4py` for now.

mcollu, 2025-04-01:
Hi, many thanks, and sorry for the delay.
I followed the following steps:
`conda install git`
`git clone https://github.com/WISDEM/WEIS.git`
`cd WEIS`
`conda env create --name weis-env -f environment.yml`
`conda activate weis-env`
`conda install -y pyoptsparse` 

At this point I get the following error - do you know what could it be?

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package libflang-5.0.0-vc14_20180207 requires vc 14, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ openfast >=4.0.1  is installable and it requires
│  └─ _openmp_mutex >=4.5 , which requires
│     └─ msys2-conda-epoch <0.0a0 , which can be installed;
├─ pin-1 is installable and it requires
│  └─ python 3.12.* , which can be installed;
├─ pyhams >=1.3  is installable with the potential options
│  ├─ pyhams 1.3.0 would require
│  │  └─ llvm-openmp >=19.1.2  with the potential options
│  │     ├─ llvm-openmp 19.1.2 would require
│  │     │  └─ openmp 19.1.2|19.1.2.* , which can be installed;
│  │     ├─ llvm-openmp 20.1.1 would require
│  │     │  └─ openmp 20.1.1|20.1.1.* , which can be installed;
│  │     ├─ llvm-openmp 19.1.3 would require
│  │     │  └─ openmp 19.1.3|19.1.3.* , which can be installed;
│  │     ├─ llvm-openmp 19.1.4 would require
│  │     │  └─ openmp 19.1.4|19.1.4.* , which can be installed;
│  │     ├─ llvm-openmp 19.1.5 would require
│  │     │  └─ openmp 19.1.5|19.1.5.* , which can be installed;
│  │     ├─ llvm-openmp 19.1.6 would require
│  │     │  └─ openmp 19.1.6|19.1.6.* , which can be installed;
│  │     ├─ llvm-openmp 19.1.7 would require
│  │     │  └─ openmp 19.1.7|19.1.7.* , which can be installed;
│  │     └─ llvm-openmp 20.1.0 would require
│  │        └─ openmp 20.1.0|20.1.0.* , which can be installed;
│  ├─ pyhams 1.3.0 would require
│  │  └─ llvm-openmp >=19.1.6 , which can be installed (as previously explained);
│  ├─ pyhams 1.3.0 would require
│  │  └─ python >=3.10,<3.11.0a0 , which conflicts with any installable versions previously reported;
│  ├─ pyhams 1.3.0 would require
│  │  └─ python >=3.11,<3.12.0a0 , which conflicts with any installable versions previously reported;
│  ├─ pyhams 1.3.0 would require
│  │  └─ m2w64-gcc-libs-core, which requires
│  │     └─ msys2-conda-epoch [20160418 |>=20160418 ], which conflicts with any installable versions ;
│  ├─ pyhams 1.3.0 would require
│  │  └─ python >=3.13,<3.14.0a0 , which conflicts with any installable versions previously reported;
│  └─ pyhams 1.3.0 would require
│     └─ python >=3.9,<3.10.0a0 , which conflicts with any installable versions previously reported;
├─ pyoptsparse is installable with the potential options
│  ├─ pyoptsparse [2.10.2|2.11.0|2.11.1|2.11.2|2.11.3] would require
│  │  └─ libflang >=5.0.0,<6.0.0.a0  with the potential options
│  │     ├─ libflang 5.0.0 would require
│  │     │  └─ vc 14 , which can be installed;
│  │     └─ libflang 5.0.0 would require
│  │        └─ openmp 5.0.0 , which conflicts with any installable versions previously reported;
│  ├─ pyoptsparse [2.10.0|2.10.1|...|2.9.4] would require
│  │  └─ python >=3.10,<3.11.0a0 , which conflicts with any installable versions previously reported;
│  ├─ pyoptsparse [2.9.0|2.9.1] would require
│  │  └─ python >=3.7,<3.8.0a0 , which conflicts with any installable versions previously reported;
│  ├─ pyoptsparse [2.10.0|2.10.1|...|2.9.4] would require
│  │  └─ python >=3.8,<3.9.0a0 , which conflicts with any installable versions previously reported;
│  ├─ pyoptsparse [2.10.0|2.10.1|...|2.9.4] would require
│  │  └─ python >=3.9,<3.10.0a0 , which conflicts with any installable versions previously reported;
│  ├─ pyoptsparse [2.10.0|2.10.1|...|2.9.4] would require
│  │  └─ python >=3.11,<3.12.0a0 , which conflicts with any installable versions previously reported;
[... 12 lines cut ...]
│  └─ pyoptsparse 2.13.0 would require
│     └─ python >=3.13,<3.14.0a0 , which conflicts with any installable versions previously reported;
└─ smt is not installable because there are no viable options
   ├─ smt [2.6.0|2.6.1|...|2.9.2] would require
   │  └─ vc >=14.2,<15 , which conflicts with any installable versions previously reported;
   ├─ smt [0.1|0.2.1|...|0.3.4] would require
   │  └─ python [2.7* |>=2.7,<2.8.0a0 ], which conflicts with any installable versions previously rep;
   ├─ smt 0.1 would require
   │  └─ python [3.5* |>=3.5,<3.6.0a0 ], which conflicts with any installable versions previously rep;
   ├─ smt 0.1 would require
   │  └─ python 3.6* , which conflicts with any installable versions previously reported;
   ├─ smt [0.1|0.2.1|...|1.0.0] would require
   │  └─ python >=3.6,<3.7.0a0 , which conflicts with any installable versions previously reported;
   ├─ smt [0.2.1|0.2.2|...|1.3.0] would require
   │  └─ python >=3.7,<3.8.0a0 , which conflicts with any installable versions previously reported;
   ├─ smt [0.5.3|0.6.0|...|2.6.3] would require
   │  └─ python >=3.8,<3.9.0a0 , which conflicts with any installable versions previously reported;
   ├─ smt [0.6.0|0.7.0|...|2.9.2] would require
   │  └─ python >=3.9,<3.10.0a0 , which conflicts with any installable versions previously reported;
   ├─ smt [1.1.0|1.2.0|...|2.9.2] would require
   │  └─ python >=3.10,<3.11.0a0 , which conflicts with any installable versions previously reported;
   ├─ smt [1.3.0|2.0.0|...|2.9.2] would require
   │  └─ python >=3.11,<3.12.0a0 , which conflicts with any installable versions previously reported;
   └─ smt [2.8.0|2.8.1|2.9.0|2.9.1|2.9.2] would require
      └─ python >=3.13,<3.14.0a0 , which conflicts with any installable versions previously reported.`

gbarter, 2025-04-02:
@mcollu - Sorry that you hit an installation snag.  I just tried on a Windows virtual machine and it seems like `pyoptsparse` is the source of the conflict.  I was able to install all the core WEIS packages in the `environment.yml` file using the `conda env create` command.  I have been using [Miniforge](https://conda-forge.org/download/), in case that helps too.

maucollu, 2025-04-04:
Many thanks - I confirm that, apart from `pyoptsparse`, all is ok. Would be glad if you can suggest an alternative to it

gbarter, 2025-04-04:
Well, the conflict with `pyoptsparse` is only on Windows, so other platforms can use that package just find.  Also, eventually the fortran compiler of the conda packages on Windows will align (the misalignment is causing the conflict now).  

WEIS-WISDEM also include a number of [optimization solvers via NLOpt](https://github.com/WISDEM/WISDEM/blob/master/wisdem/optimization_drivers/nlopt_driver.py).  We have been using these more and more over pyOptSparse.