NLRWindSystems/WEIS #367: Installing WEIS on Kestrel (README.md)

yonghoonlee, 2025-03-21:
## Description
Kestrel module names are not matching with the README.md file.
The loading commands for the necessary modules should be replaced by:
```
module load intel-oneapi-compilers intel-oneapi-mpi intel-oneapi-mkl
```

### Steps to reproduce issue
N/A

### Current behavior
N/A

### Expected behavior
N/A

### Code versions
N/A

ptrbortolotti, 2025-03-25:
It's true that the `README.md` file is outdated on both `master` and `develop` branch, but I am not sure about this list of modules

in my profile I load `module load tmux intel-oneapi-mkl/2023.2.0-intel mamba` and as I result I get 

```
Currently Loaded Modules:
  1) tmux/3.3a                          3) intel-oneapi-mkl/2023.2.0-intel
  2) intel-oneapi-tbb/2021.10.0-intel   4) mamba/1.4.2

ptrbortolotti, 2025-03-25:
@dzalkind any guidance from the WEIS master?

dzalkind, 2025-03-26:
I don't have any strong preferences here.  These modules are always changing and often result in similar lists.

@mayankchetan, I usually ask for your input on these things.  What's worked well for you recently in the downwind project?

yonghoonlee, 2025-03-26:
All,
I do not have any preference here. Just wanted to let you know the lists are different to the current names of modules available. If no changes should be made, anyone can close the issue. Thank you.
Yong Hoon

mayankchetan, 2025-03-26:
I think the list maintained here should be as generic as possible (avoid pinning versions). Once users get familiar with Kestrel, they tend to have three own preferred list that they use (I'm guilt of it).

yonghoonlee, 2025-03-27:
Thank you for your inputs. I will close the issue.