NLRWindSystems/WEIS #368: Schema validation issue

yonghoonlee, 2025-03-21:
## Description
There is a schema validation issue occurring when running the example 15 with OpenFAST script.

### Steps to reproduce issue

1. Install the latest version of `develop` branch of the WEIS
2. Go to example 15 directory
3. Run `weis_driver_OpenFAST.py` script

### Current behavior
```
cd examples/15_RAFT_Studies
python weis_driver_OpenFAST.py
```

```
Using weis.aeroelasticse in rosco.toolbox...
Traceback (most recent call last):
  File "/kfs2/projects/weis/yonghoonlee/WEIS/examples/15_RAFT_Studies/weis_driver_OpenFAST.py", line 11, in <module>
    wt_opt, modeling_options, opt_options = weis_main(fname_wt_input, fname_modeling_options, fname_analysis_options)
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/kfs2/projects/weis/yonghoonlee/WEIS/weis/main.py", line 48, in weis_main
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, 
                                            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/kfs2/projects/weis/yonghoonlee/WEIS/weis/glue_code/runWEIS.py", line 24, in run_weis
    wt_initial = WindTurbineOntologyPythonWEIS(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/kfs2/projects/weis/yonghoonlee/WEIS/weis/glue_code/gc_LoadInputs.py", line 42, in __init__
    self.wt_init          = sch.load_geometry_yaml(fname_input_wt)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/kfs2/projects/weis/yonghoonlee/WEIS/weis/inputs/validation.py", line 35, in load_geometry_yaml
    return _validate(finput, merged_schema, defaults=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yonghoonlee/miniforge3/envs/weis-env/lib/python3.12/site-packages/wisdem/inputs/validation.py", line 194, in _validate
    validator(schema_dict).validate(input_dict)
  File "/home/yonghoonlee/miniforge3/envs/weis-env/lib/python3.12/site-packages/jsonschema/validators.py", line 451, in validate
    raise error
jsonschema.exceptions.ValidationError: 0.0 is not of type 'array'

Failed validating 'type' in schema['properties']['components']['properties']['floating_platform']['properties']['members']['items']['properties']['Ca']:
    {'description': 'User-defined added mass coefficient if axi-symmetric '
                    'or user-defined added mass coefficient in x-direction',
     'type': 'array',
     'default': [-1.0],
     'items': {'type': 'number', 'minimum': -1.0}}

On instance['components']['floating_platform']['members'][8]['Ca']:
    0.0
```

### Expected behavior

### Code versions
```
conda list
```

```
# packages in environment at /home/yonghoonlee/miniforge3/envs/weis-env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                  3_kmp_llvm    conda-forge
_x86_64-microarch-level   4               2_sapphirerapids    conda-forge
alabaster                 1.0.0              pyhd8ed1ab_1    conda-forge
alsa-lib                  1.2.13               hb9d3cd8_0    conda-forge
ampl-asl                  1.0.0                h5888daf_2    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     25.3.0             pyh71513ae_0    conda-forge
babel                     2.17.0             pyhd8ed1ab_0    conda-forge
[... 284 lines cut ...]
xcb-util-keysyms          0.4.1                hb711507_0    conda-forge
xcb-util-renderutil       0.3.10               hb711507_0    conda-forge
xcb-util-wm               0.4.2                hb711507_0    conda-forge
xkeyboard-config          2.43                 hb9d3cd8_0    conda-forge
xorg-libice               1.1.2                hb9d3cd8_0    conda-forge
xorg-libsm                1.2.6                he73a12e_0    conda-forge
xorg-libx11               1.8.12               h4f16b4b_0    conda-forge
xorg-libxau               1.0.12               hb9d3cd8_0    conda-forge
xorg-libxcomposite        0.4.6                hb9d3cd8_2    conda-forge
xorg-libxcursor           1.2.3                hb9d3cd8_0    conda-forge
xorg-libxdamage           1.1.6                hb9d3cd8_0    conda-forge
xorg-libxdmcp             1.1.5                hb9d3cd8_0    conda-forge
xorg-libxext              1.3.6                hb9d3cd8_0    conda-forge
xorg-libxfixes            6.0.1                hb9d3cd8_0    conda-forge
xorg-libxi                1.8.2                hb9d3cd8_0    conda-forge
xorg-libxrandr            1.5.4                hb9d3cd8_0    conda-forge
xorg-libxrender           0.9.12               hb9d3cd8_0    conda-forge
xorg-libxtst              1.2.5                hb9d3cd8_3    conda-forge
xorg-libxxf86vm           1.1.6                hb9d3cd8_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.5                h3b0a872_7    conda-forge
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zstandard                 0.23.0          py312h66e93f0_1    conda-forge
zstd                      1.5.7                hb8e6e7a_2    conda-forge
```

dzalkind, 2025-03-21:
Hi YongHoon,

This example is not tested and is likely to be removed from the examples.  I recommend pointing to a [maintained geometry file](https://github.com/WISDEM/WEIS/tree/main/examples/00_setup/ref_turbines).

Best, Dan

yonghoonlee, 2025-03-26:
@dzalkind Thank you for letting me know. I will close the issue.

closed 2025-03-26 by yonghoonlee