NLRWindSystems/WEIS #369: Aerodynamic and structural optimization of blades using WEIS

EVElove444, 2025-03-22:
Please, I want to use WEIS for optimization of torsion angle chord length and layup thickness of a blade using a multi-objective optimization algorithm, which case should I run?

ptrbortolotti, 2025-03-23:
hello, your request raises a number of challenges. Nothing insurmountable, but you will need to become familiar with WEIS to get what you need

Indeed, with PR #362 we've just added torsional stiffness to OpenFAST, and we don't have anything like you're asking ready to share

To get what you want, you can start from the WISDEM blade optimization [example 3a](https://wisdem.readthedocs.io/en/master/examples/03_blade/tutorial.html) and take that [aerostrctural analysis options file](https://github.com/WISDEM/WISDEM/blob/master/examples/03_blade/analysis_options_aerostruct.yaml)

Next, in the modeling options you will need to call the OpenFAST module BeamDyn, like [here](https://github.com/WISDEM/WEIS/blob/a29827c9051506018895f57202e7cb366bdd76d6/examples/06_IEA-15-240-RWT/modeling_options.yaml#L41)

The trickier aspects will be:
- Adding the statistics of the torsion angle among the OpenMDAO outputs so that you can constrain/minimize them. You can do so by working [here](https://github.com/WISDEM/WEIS/blob/a32208f04e241da38e2a9e67c069939cb6f1f5a6/weis/aeroelasticse/openmdao_openfast.py#L2212)
- BeamDyn still makes OpenFAST very slow. OpenFAST 5.0 will resolve that, but it's planned to be released no earlier than the summer of 2025
- I have not run multi-objective optimizations in WISDEM, at least not recently. Googling it suggests that OpenMDAO may support it, but that's again something you will need to explore. You could start [here](https://stackoverflow.com/questions/35058323/pareto-frontier-generation-for-multi-objective-prob-using-openmdao-1-x)

EVElove444, 2025-03-23:
@ptrbortolotti I would like to optimize the blade's twist angle, chord length, and composite layup using a multi-objective optimization algorithm. How should I modify the code accordingly? Furthermore, if I want to include failure constraints based on the Tsai-Wu failure criterion, what changes are required in the code?

ptrbortolotti, 2025-03-24:
please familiarize yourself with WISDEM example 3a. I am not too sure how to do multi-objective optimization in OpenMDAO, a quick search shows me that OpenMDAO does not support it natively. for Tsai-Wu, I don't think WISDEM has the right fidelity to compute it accurately. for that I'd recommend some higher fidelity blade structural solver such as [pyNuMAD](https://github.com/sandialabs/pyNuMAD), although pyNuMAD is not tailored to optimization

closed 2025-04-01 by ptrbortolotti