NLRWindSystems/WEIS #373: Current examples do not run for OpenFAST analysis time more than 1 second

yonghoonlee, 2025-04-09:
## Description

The current IEA15MW turbine examples do not work for OpenFAST with larger than 1 second of simulation.
I tried different DLCs with simulation time of 600 seconds, and none of them works.
I also tried Tower_DV case, UMaine Semi case, etc, and none of them works.

### Steps to reproduce issue

1. Change `modeling_options` to run OpenFAST more than 1 second for DLC 1.1

```
diff --git a/examples/06_IEA-15-240-RWT/modeling_options_umaine_semi.yaml b/examples/06_IEA-15-240-RWT/modeling_options_umaine_semi.yaml
index 04c0e491..f21eecc7 100644
--- a/examples/06_IEA-15-240-RWT/modeling_options_umaine_semi.yaml
+++ b/examples/06_IEA-15-240-RWT/modeling_options_umaine_semi.yaml
@@ -105,6 +105,14 @@ DLC_driver:
+        - DLC: "1.1"
+          ws_bin_size: 2
+          wind_speed: [14.]
+          wave_height: [7.]
+          wave_period: [1.]
+          n_seeds: 1
+          analysis_time: 600.   # These inputs control the simulation length
+          transient_time: 120.    # These inputs control the simulation length
@@ -120,10 +128,10 @@ DLC_driver:
-        - DLC: "1.6"
-          wind_speed: [12]
-          analysis_time: 1.
-          transient_time: 0.
```

2. Run weis driver

```
python weis_driver_umaine_semi.py
```

### Current behavior

OpenFAST does not simulate time-domain problem correctly

```

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2025 National Renewable Energy Laboratory
 Copyright (C) 2025 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST-v4.0.2
 Compile Info:
  - Compiler: GCC version 13.2.0
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: No
  - Date: Feb 14 2025
  - Time: 23:40:15
 Execution Info:
  - Date: 04/09/2025
  - Time: 11:16:44-0500

 OpenFAST input file heading:
     Generated with OpenFAST_IO

  Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
  Running InflowWind.

    Reading a 25x25 grid (300 m wide, 5.18799E-04 m to 300 m above ground) with a characteristic
    wind speed of 10 m/s. This full-field file was generated by TurbSim (from OpenFAST-v4.0.2) on
    09-Apr-2025 at 10:01:40.

    Processed 10 time steps of 20-Hz grid-field data (period of 0.5 seconds).
  Running SeaState.
  Generating incident wave kinematics and current time history.
  Running AeroDyn.
 AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
 Skipping nodal outputs.
    AeroDyn: projMod: 1
    Projection: legacy (NoSweepPitchTwist), BEM: legacy (2D)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 1, blade 1)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 2, blade 1)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 1, blade 2)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 2, blade 2)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 1, blade 3)
 Warning: Turning off Unsteady Aerodynamics because polar has constant data. (node 2, blade 3)
  Running HydroDyn.
  Reading in WAMIT output with root name
  "/Users/yhlee/work/WEIS-FTW/examples/01_aeroelasticse/OpenFAST_models/IEA-15-240-RWT/IEA-15-240-R
  WT-UMaineSemi/HydroData/IEA-15-240-RWT-UMaineSemi".
  Computing radiation impulse response functions and wave diffraction forces.
  Running MoorDyn (v2.2.2, 2024-01-16).
    This is MoorDyn v2, with significant input file changes from v1.

 **************************************************************************************************
 MoorDyn

 Copyright (C) 2025 National Renewable Energy Laboratory
 Copyright (C) 2025 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

    Parsing MoorDyn input file: ./DLC1.1_0_weis_job_0_MoorDyn.dat
   Created mooring system:  3 lines, 6 points, 0 rods, 0 bodies.
    Finalizing initial conditions using dynamic relaxation.

   t=1  FairTen 1: 1.01418E+07, 9.87305E+06, 9.76892E+06                                          
    Fairlead tensions did not converge within TMaxIC=1 seconds.
    MoorDyn initialization completed.
  Running ServoDyn.
  Running ServoDyn Interface for Bladed Controllers (using GNU Fortran for Linux).
 Using legacy Bladed DLL interface.

 FAST_InitializeAll:ED_Init:ED_ReadInput:ReadBladeInputs:ReadBladeFile:The ElastoDyn Blade file,
 ./DLC1.1_0_weis_job_0_ElastoDynBlade.dat, DISTRIBUTED BLADE PROPERTIES table contains the
 PitchAxis column.  This column is unused and will be removed in future releases
 FAST_InitializeAll:InitModuleMappings:ED_2_AD_BladeMotion(1):MeshMapCreate:CreateMotionMap_L2_to_L
 2:CreateMapping_ProjectToLine2:Found close value for node 60. (3.03409E-05 m)

  Time: 0 of 0.5 seconds.
                                                                               
------------------------------------------------------------------------------
Running ROSCO-v2.9.0
A wind turbine controller framework for public use in the scientific field    
Developed in collaboration: National Renewable Energy Laboratory              
                            Delft University of Technology, The Netherlands   
------------------------------------------------------------------------------
Generator speed:    7.1 RPM, Pitch angle:   0.0 deg, Power:     0.0 kW, Est. wind Speed:  10.2 m/s
 FAST_Solution:FAST_UpdateStates:FAST_AdvanceStates:SrvD_UpdateStates:DLL_controller_call:BladedInt
 erface option was designed for an explicit-loose coupling scheme. Using last calculated values
 from DLL on all subsequent calls until time is advanced. Warning will not be displayed again.
 FAST_Solution:FAST_UpdateStates:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcO
 utput:BEMT_CalcOutput(node 3, blade 1):UA_CalcOutput:UA_BlendSteady:Temporarily turning off UA
 due to high angle of attack or low relative velocity. This warning will not be repeated though
 the condition may persist.
 FAST_Solution:FAST_UpdateStates:FAST_AdvanceStates:AD_UpdateStates:BEMT_UpdateStates(node 1,
 blade 1):ComputeTau:Rotor-averaged axial induction factor is greater than 0.5; limiting
 time-varying tau1. This message will not be repeated though the condition may persist.
                                                                                                  
  Total Real Time:       0.843 seconds
  Total CPU Time:        0.81442 seconds
  Simulation CPU Time:   0.38249 seconds
  Simulated Time:        0.5 seconds
  Time Ratio (Sim/CPU):  1.3072

  OpenFAST terminated normally.

Runtime: 	DLC1.1_0_weis_job_0.fst = 0.88  s
RuntimeWarning: /Users/yhlee/miniforge3/envs/weis-ftw-dev/lib/python3.12/site-packages/pCrunch/fatigue.py:329
invalid value encountered in powerRuntimeWarning: /Users/yhlee/miniforge3/envs/weis-ftw-dev/lib/python3.12/site-packages/fatpack/endurance.py:207
invalid value encountered in powerRuntimeWarning: /Users/yhlee/miniforge3/envs/weis-ftw-dev/lib/python3.12/site-packages/pCrunch/fatigue.py:326
divide by zero encountered in scalar divideWARNING: DLC 1.1 is being used for AEP calculations.  Use the AEP DLC for more accurate wind modeling with constant TI.
WARNING: OpenFAST is run at a single wind speed. AEP cannot be estimated. Using average power instead.
```

### Code versions

Latested version (WEIS 1.5.2 release) fresh installed.

dzalkind, 2025-04-09:
Hi @yonghoonlee,

We recently added a test mode to the examples for speeding up the tests: https://github.com/WISDEM/WEIS/blob/86a5df1b8792a3bf036642d1b2bd3557ace7f555/examples/02_run_openfast_cases/weis_driver_loads.py#L4

Set that flag to `False` and it should run the full example.  

This should be documented or automatically toggled when running in the CI.  Keeping this open until then.

Best, Dan

gbarter, 2025-04-09:
This may be related to a couple of Issues that have flared up on the IEA15 repository this week. . .

gbarter, 2025-04-09:
Sorry, I was not able to reproduce the error.  Did Dan's tip help you solve the issue?

yonghoonlee, 2025-04-10:
@gbarter Yes, Dan's tip resolved the issue. Thank you. As Dan asked, I am keeping this issue open, but if needed, please feel free to close this issue. Thank you.

closed 2025-05-14 by dzalkind