NLRWindSystems/WEIS #374: Overriding Control dict in WindIO yaml

TheMercer, 2025-04-15:
Hi,

I been having an issue where I have to check the yaml to change the TSR in wisdem run from WEIS, what's the best way in WEIS inputs to override: TSR, max tip speed,  etc etc?

gbarter, 2025-04-16:
Have you looked in the `control` section of the input yaml file?  [Here](https://github.com/WISDEM/WEIS/blob/main/examples/00_setup/ref_turbines/IEA-15-240-RWT.yaml) is the 15MW section.

TheMercer, 2025-04-16:
Yeah how do i override it from WEIS as when i set the TSR in wisdem. I get the yaml tasr. overriding it

gbarter, 2025-04-16:
You'll have to explain more what you mean by "override".  The TSR generally only applies in Region II of the power curve.  At cut in, near-rated, and above-rated, other control laws will apply.

closed 2025-04-17 by TheMercer