NLRWindSystems/WEIS #375: Blade scaling stiffness

TheMercer, 2025-04-15:
I have been trying to scale a turbine and I made spar and TE to be varied however i don't see them changing much. 

I think this looks to my model having lower blade stiffness properties. 

What's the recommendations to getting airfoil thickness optimization? (Also not staying basically the same)

ptrbortolotti, 2025-04-17:
Hi @TheMercer,
you can either adjust the spanwise positions of your airfoils in your geometry yaml file [here](https://github.com/WISDEM/WISDEM/blob/14ad23788496bc130d69d613b9c408f9615f21b7/examples/03_blade/BAR_USC.yaml#L8), making sure that you don't have the `rthick` distribution defined in [`outer_shape_bem`](https://github.com/WISDEM/WISDEM/blob/14ad23788496bc130d69d613b9c408f9615f21b7/examples/03_blade/BAR_USC.yaml#L6) (if you do, just comment it out)

or you can let the optimizer adjust those positions. For the latter approach, in the analysis options yaml, `design_variables` -> `blade` -> `aero_shape` [link](https://github.com/WISDEM/WISDEM/blob/14ad23788496bc130d69d613b9c408f9615f21b7/examples/03_blade/analysis_options_aero.yaml#L12), add these lines (where `af_start` controls the first airfoil along span that can be moved by the optimizer.

```yaml
            af_positions:
                flag: True
                af_start: 4

TheMercer, 2025-04-17:
Ooo yeah, thats a good point. I been getting to change the airfoil position

I will try this

closed 2025-05-04 by TheMercer