NLRWindSystems/WEIS #380: Inquiry on Setting Up Aero-Elastic-Control Multi-Objective Optimization in WEIS

EVElove444, 2025-04-26:
I hope you’re doing well. I am a researcher at [Your Institution], and I am interested in using WEIS to perform a multi-objective optimization of a wind turbine blade that simultaneously considers:

Aerodynamic variables: spanwise twist and chord distributions

Structural variables: spar cap thickness (laminate stacking)

Control parameters: ROSCO pitch-control gains (e.g. natural frequency ω and damping ζ)

My current understanding of the workflow is as follows:

Design Variable Initialization
I register twist, chord, spar cap thickness, and ROSCO controller gains as design variables in analysis_options.yaml (or via an OpenMDAO script) under the pyOptSparse/NSGA-II driver.

WISDEM Blade Modeling
At each iteration, RotorSE rebuilds the blade geometry and structural model from the new twist/chord/laminate inputs and computes blade mass.

OpenFAST Input Generation
The gluecode module writes the updated blade geometry, structural properties, and controller parameters into the OpenFAST input files within WEIS.

Aeroelastic-Control Simulation
The aeroelasticse subsystem invokes OpenFAST (with the ROSCO controller) to simulate the coupled aero-elastic behavior and control response, producing time-domain outputs such as power, loads, and tip deflection.

Post-Processing Metrics
From the OpenFAST results I calculate annual energy production (AEP) using the power curve and wind-speed distribution, and extract blade mass from WISDEM outputs.

Optimization Loop
NSGA-II performs non-dominated sorting on the two objectives (maximize AEP, minimize blade mass) to generate the next generation of design variables until convergence.

Could you please confirm whether this workflow aligns with WEIS’s recommended approach? In particular:

Are there example templates or best practices in modeling_options.yaml and analysis_options.yaml for jointly optimizing blade geometry, layup thickness, and ROSCO controller settings?

Do I need to enable or adjust any additional flags (e.g. under control, gluecode, or aeroelasticse) to ensure the blade and controller variables are correctly passed through each subsystem?

Any pointers to documentation snippets or sample cases would be greatly appreciated.

Thank you very much for your time and assistance. I look forward to your guidance.

gbarter, 2025-04-27:
The answer here is the same as [your question on the WISDEM Issues](https://github.com/WISDEM/WISDEM/issues/625).  Your rough outline of the WEIS workflow is accurate, but we don't have a multi-objective template for you nor have we used WEIS with NSGA-II.  You will have to blaze your own trail.

EVElove444, 2025-04-27:
@gbarter Currently if I use a single objective, can WEIS currently achieve blade shape initiation and structural optimization + control optimization for consideration of air bombs

EVElove444, 2025-04-27:
@gbarter Currently, if I use a single‐objective optimization, is it possible in WEIS to perform aero‐elastic blade shape and structural optimization together with control optimization?

gbarter, 2025-04-28:
Multi-disciplinary design optimization with the controller as part of the design variables is the primary mission of WEIS, so what you described is certainly doable (but not necessarily easy as it can be a noisy tradespace).

dzalkind, 2025-05-14:
Please let us know if you have additional questions or would like support setting up new workflows.