NLRWindSystems/WEIS #384: Directly obtain the required hydrodynamic data from a given floating platform geometry?

fangjianju, 2025-05-09:
I have a question about WEIS: In WEIS, is it possible to directly obtain the required hydrodynamic data from a given floating platform geometry? How is the mesh for the hydrodynamic solver automatically generated?

gbarter, 2025-05-09:
Hello @fangjianju - What do you have in mind when you say "required hydrodynamic data".  If pyHAMS is invoked for the WEIS run, the input and output files should be present when the simulation is complete.  I have asked others to chime in on your meshing question.

yqliaohk, 2025-05-09:
Hi @fangjianju - 
Agree with @gbarter - some clarifications on the hydrodynamic data can help us better answer.
For meshing, we have two approaches: (1) we have internal functions to generate meshes for circular and rectangular members, but this is only limited to non-intersected members (can be multiple members); (2) we have another approach that relies on external packages such as pygmsh and meshmagick to generate meshes for more complex geometries that have intersected members. This capability will be in RAFT release 1.4. You can choose either approach by adjusting the option `intersection_mesh`.

closed 2025-07-31 by dzalkind