NLRWindSystems/WEIS #385: Error calling compute(), 'numpy.float64' object is not iterable

mihopark, 2025-05-12:
When I run the example(06_IEA-15-240-RWT/weis_driver_tower_DVs.py) below, I get the following error. Please tell me how to fix it.

  File "/home//anaconda3/envs/weis152/lib/python3.12/site-packages/openfast_io/FAST_writer.py", line 2361, in write_MoorDyn ln.append('|'.join([float_default_out(a, trim=True) for a in self.fst_vt['MoorDyn']['EA'][i]]))
                                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
TypeError: 'aeroelastic' <class FASTLoadCases>: Error calling compute(), 'numpy.float64' object is not iterable

Thank you for your advice.

gbarter, 2025-05-12:
I encountered this error today as well.  I will let you know when I track it down.

gbarter, 2025-05-13:
Please check the version number of you `openfast_io` package (with `conda list`), and make sure it is up-to-date with the latest version, v4.0.4

bluemiracle4862, 2025-05-13:
I checked the openfast_io of version v4.0.4 installed.

Thank you for advice.

gbarter, 2025-05-13:
So, the problem persists?  Are you using the `master` branch or `develop` branch of WEIS?

bluemiracle4862, 2025-05-13:
I'm using master branch of WEIS.
Thank you for advice.

maucollu, 2025-05-14:
HI, just to let you know that I have the same issue with the  weis_driver_umaine_semi.py in example 06.
I am using the openfast_io v4.0.4, main branch.

dzalkind, 2025-05-14:
Hi,

Thanks for bringing this to out attention.  It seems like we have not kept up to date with the `openfast-io` dependency. 

Please try `pip install openfast-io==4.0.2`.  We'll sort out the issue with v4.0.4 in the meantime.

Best, Dan

maucollu, 2025-05-16:
Thanks, @dzalkind