NLRWindSystems/WEIS #386: Why is Blade stiffness GJ much lower then expected

TheMercer, 2025-05-12:
```
    blade:
        aero_shape:
            twist:
                flag: True             # Flag to optimize the twist
                inverse: False         # Flag to determine twist from the user-defined desired margin to stall (defined in constraints)
                n_opt: 8               # Number of control points along blade span
                max_decrease: 0.08722222222222221 # Maximum decrease for the twist in [rad] at the n_opt locations
                max_increase: 0.08722222222222221 # Maximum increase for the twist in [rad] at the n_opt locations
                index_start: 0         # Lock the first two DVs from blade root
                index_end: 8           # All DVs close to blade tip are active
            chord:
                flag: True             # Flag to optimize the chord
                n_opt: 8               # Number of control points along blade span
                max_decrease: 0.3      # Minimum multiplicative gain on existing chord at the n_opt locations
                max_increase: 3.       # Maximum multiplicative gain on existing chord at the n_opt locations
                index_start: 0         # Lock the first DV from blade root
                index_end: 8           # The last DV at blade tip is locked
            af_positions:
                flag: True
                af_start: 1
        structure:
            - layer_name: Spar_Cap_SS
              n_opt: 8               # Number of control points along blade span
              max_decrease: 0      # Maximum nondimensional decrease at the n_opt locations
              max_increase: 5      # Maximum nondimensional increase at the n_opt locations
              index_start: 0         # Lock the first DV from blade root
              index_end: 8           # The last DV at blade tip is locked
            - layer_name: Spar_Cap_PS
              n_opt: 8               # Number of control points along blade span
              max_decrease: 0      # Maximum nondimensional decrease at the n_opt locations
              max_increase: 5      # Maximum nondimensional increase at the n_opt locations
              index_start: 0         # Lock the first DV from blade root
              index_end: 8           # The last DV at blade tip is locked
            - layer_name: LE_reinforcement
              n_opt: 8               # Number of control points along blade span
              max_decrease: 0      # Maximum nondimensional decrease at the n_opt locations
              max_increase: 5.       # Maximum nondimensional increase at the n_opt locations
              index_start: 0         # Lock the first DV from blade root
              index_end: 8           # The last DV at blade tip is locked
            - layer_name: TE_reinforcement_SS
              n_opt: 8               # Number of control points along blade span
              max_decrease: 0      # Maximum nondimensional decrease at the n_opt locations
              max_increase: 5.       # Maximum nondimensional increase at the n_opt locations
              index_start: 0         # Lock the first DV from blade root
              index_end: 8           # The last DV at blade tip is locked
            - layer_name: TE_reinforcement_PS
              n_opt: 8               # Number of control points along blade span
              max_decrease: 0      # Maximum nondimensional decrease at the n_opt locations
              max_increase: 5.       # Maximum nondimensional increase at the n_opt locations
              index_start: 0         # Lock the first DV from blade root
              index_end: 8           # The last DV at blade tip is locked
            - layer_name: Shell_skin
              n_opt: 8               # Number of control points along blade span
              max_decrease: 0      # Maximum nondimensional decrease at the n_opt locations
              max_increase: 5.       # Maximum nondimensional increase at the n_opt locations
              index_start: 0         # Lock the first DV from blade root
              index_end: 8           # The last DV at blade tip is locked
            - layer_name: Shell_skin_inner
              n_opt: 8               # Number of control points along blade span
              max_decrease: 0      # Maximum nondimensional decrease at the n_opt locations
              max_increase: 5.       # Maximum nondimensional increase at the n_opt locations
              index_start: 0         # Lock the first DV from blade root
              index_end: 8           # The last DV at blade tip is locked
merit_figure: LCOE
    
constraints:
    blade:
        strains_spar_cap_ss:
            flag: True         # Flag to impose constraints on maximum strains (absolute value) in the spar cap on the blade suction side
            max:    3500.e-6   # Value of maximum strains [-]
            index_start: 1     # Do not enforce constraint at the first station from blade root of the n_opt from spar_cap_ss
            index_end: 8       # Do not enforce constraint at the last station at blade tip of the n_opt from spar_cap_ss
        strains_spar_cap_ps:
            flag: True         # Flag to impose constraints on maximum strains (absolute value) in the spar cap on the blade pressure side
            max:    3500.e-6   # Value of maximum strains [-]
            index_start: 1     # Do not enforce constraint at the first station from blade root of the n_opt from spar_cap_ps
            index_end: 8       # Do not enforce constraint at the last station at blade tip of the n_opt from spar_cap_ps
        strains_te_ss:
            flag: False        # Flag to impose constraints on maximum strains (absolute value) in the spar cap on the blade suction side
            max:    3500.e-6   # Value of maximum strains [-]
            index_start: 1     # Do not enforce constraint at the first station from blade root of the n_opt from spar_cap_ss
            index_end: 8       # Do not enforce constraint at the last station at blade tip of the n_opt from spar_cap_ss
        strains_te_ps:
            flag: False        # Flag to impose constraints on maximum strains (absolute value) in the spar cap on the blade pressure side
            max:    3500.e-6   # Value of maximum strains [-]
            index_start: 1     # Do not enforce constraint at the first station from blade root of the n_opt from spar_cap_ps
            index_end: 8       # Do not enforce constraint at the last station at blade tip of the n_opt from spar_cap_ps
        tip_deflection:
            flag: True
            margin: 1.4175
        stall:
            flag: True    # Constraint on minimum stall margin
            margin: 0.087 # Value of minimum stall margin in [rad]
        moment_coefficient:
            flag: True    # Constraint on minimum stall margin
            max: 0.16
```

ptrbortolotti, 2025-05-12:
@TheMercer, is the issue after the optimization, or at the starting point? We've observed that PreComp underestimates GJ, and that was also described here https://github.com/WISDEM/WEIS/pull/362

TheMercer, 2025-05-12:
Well I am starting with the IEA 15 MW blade. So I assume the GJ is fine. 

Then after the optimisation when I get the new WindIO file. I can see Precomp predicts low and SONATA for the same file also the GJ is less than IEA 15MW blade (I was expecting around or stiffer)

TheMercer, 2025-05-12:
The stiffness near the root seems "fine" however the GJ as you go along the span deltas more

TheMercer, 2025-05-12:
[image]

TheMercer, 2025-05-12:
Due to the stiffness being much lower as we go along the span. When running this in OrcaFlex, the blade fails due to the blade twisting under load and causing the gen speed to go unstable.

TheMercer, 2025-05-12:
Weis opt GJ |  | iea 15mw
-- | -- | --
7.77E+07 |   | 8.75E+07
7.35E+07 |   | 8.33E+07
6.24E+07 |   | 6.99E+07
4.77E+07 |   | 5.41E+07
3.22E+07 |   | 4.15E+07
2.20E+07 |   | 3.06E+07
1.51E+07 |   | 2.19E+07
9.51E+06 |   | 1.50E+07
5.88E+06 |   | 8.06E+06
4.07E+06 |   | 5.94E+06
2.92E+06 |   | 3.82E+06
1.79E+06 |   | 2.41E+06
6.58E+05 |   | 1.72E+06
1.03E+05 |   | 1.02E+06
3.81E+04 |   | 8.60E+05
3.57E+04 |   | 7.02E+05
3.29E+04 |   | 5.90E+05
2.97E+04 |   | 5.24E+05
2.65E+04 |   | 4.58E+05
2.35E+04 |   | 4.20E+05
2.08E+04 |   | 3.82E+05
1.86E+04 |   | 3.47E+05
1.65E+04 |   | 3.15E+05
1.47E+04 |   | 2.84E+05
1.30E+04 |   | 2.58E+05
1.15E+04 |   | 2.32E+05
1.03E+04 |   | 2.09E+05
9.11E+03 |   | 1.88E+05
8.07E+03 |   | 1.67E+05
7.13E+03 |   | 1.51E+05
6.29E+03 |   | 1.35E+05
5.53E+03 |   | 1.20E+05
4.76E+03 |   | 1.07E+05
3.97E+03 |   | 9.36E+04
3.19E+03 |   | 8.40E+04
2.51E+03 |   | 7.43E+04
1.98E+03 |   | 6.56E+04
1.55E+03 |   | 5.81E+04
1.18E+03 |   | 5.05E+04
8.59E+02 |   | 4.48E+04
6.04E+02 |   | 3.91E+04
4.20E+02 |   | 3.35E+04
3.09E+02 |   | 2.80E+04
2.56E+02 |   | 2.25E+04
2.25E+02 |   | 1.87E+04
1.92E+02 |   | 1.49E+04
1.48E+02 |   | 1.17E+04
1.00E+02 |   | 9100.9
5.82E+01 |   | 6500.8
2.64E+01 |   | 3924
6.64E+00 |   | 1347.2
3.45E-02 |   | 58.788

TheMercer, 2025-05-12:
the layers used etc are from this post. To run it directly some edit are needed and I can update the latest I have later today 
https://github.com/WISDEM/SONATA/issues/26

TheMercer, 2025-05-12:
I checked the thickness on each layer, pretty much the same as IEA 15MW. So only thing now I can think is some processing is somehow different or the start and end arc is messed up to the point is messing this calc up

ptrbortolotti, 2025-05-12:
you have multiple things going on here... the first one is that PreComp generally underestimates GJ with respect to ANBA/VABS. So even for a given design, you will see differences whether you run WEIS or SONATA

Next, you are running an optimization, and the structural problem may not be nicely posed. you are asking to minimize LCOE and you are giving many degrees of freedom to the optimizer to reduce layer thickness, hence blade mass, blade cost, and LCOE. you do list some constraints, but these are possibly no sufficient to limit the unrealistic drop in GJ. also, you should probably check whether your optimizer is converging anywhere.

I would recommend stepping back to simpler optimization examples before jumping all in into a full LCOE optimization. The WISDEM examples can be of help

TheMercer, 2025-05-12:
Oh okay, so would you recommend doing the Basic inverse design one ? 

Shouldn't the thickness never decrease as I set it to be max decrease 0 and max increase 5 ?

ptrbortolotti, 2025-05-12:
Oh I see, that bound is multiplicative, you need to set it to 1 to prevent a layer thickness to drop

TheMercer, 2025-05-12:
Ohhhh that makes sense, I will try that while I also reduce the problem 😂😂. Feel dumb haha

TheMercer, 2025-05-12:
I don't think thickness changed that much when I plotted them between idea 15mw and this one. Somehow the stiffness is affect by the thickness looking the same or similar

ptrbortolotti, 2025-05-12:
Your chord may also be dropping, which reduces GJ

TheMercer, 2025-05-12:
Yeah, that's true. I will look at that also

TheMercer, 2025-05-12:
So if I was to try and scale to a different MW, provide initial chord and twist and probably set the multipler

1.2 to 1.5 for example so it does shift the stiffness up at a base ?

ptrbortolotti, 2025-05-12:
yes, that will push the initial thickness up 20/50%

TheMercer, 2025-05-12:
Thank you for your support again

closed 2025-05-12 by TheMercer