NLRWindSystems/WEIS #390: How to increase the number of cores with 02_generate_openfast_model_for_dlcs example

bluemiracle4862, 2025-05-14:
Hello.

I have another question with 02_generate_openfast_model_for_dlcs/iea15_monopile_modeling.yaml

How do I increase the cores ?

I try it with the following input but It didn's work.

---------------------------------
    openfast_configuration:
        use_exe: True
        cores: 8
        mpi_run: True
---------------------------------

Thank you for your advice.

dzalkind, 2025-05-14:
Hi,

Here is some documentation about running WEIS in parallel: https://weis.readthedocs.io/en/docs/running_weis.html#running-weis-in-parallel

Best, Dan

dzalkind, 2025-05-14:
@ptrbortolotti, it seems like we're setting `cores` and `mpi_run` automatically now.  Can you confirm that I'm understanding that correctly?

If so, I'll remove them from the input schema for clarity.

ptrbortolotti, 2025-05-14:
I think you are good to remove them from the schema

bluemiracle4862, 2025-05-23:
Thank you for your advice.

briantempleton, 2025-06-05:
Thank you for your quick response! Answers for #3 make sense.

I am still thinking on #1/#2 related to MPI and not seeing evidence of parallel utilization during the OpenFAST run component. Upfront the only thing I could think of is that I am running on a low number of cores (currently have 2 allocated on this system)--but also tried running on 4 "cores" using  `--map-by :OVERSUBSCRIBE` in case there are "primary" processes consuming some of the MPI workers. Didn't seem to help.

To answer your questions:
a) I do have mpi4py installed in my conda environment. (I followed the instructions in the main readme on this git site, taking note of specified Linux specific items, e.g., mpi4py.)
b) I do get an mpi object <module 'mpi4py.MPI' from '`[home dir]`/miniconda3/envs/weis-env3/lib/python3.12/site-packages/mpi4py/MPI.cpython-312-x86_64-linux-gnu.so'> for MPI when it loaded in WEIS/weis/main.py when using the following call:
     _mpiexec -np 2 python weis_driver_loads.py_.
Additionally, I see related console outputs (e.g., "All MPI subranks closed.")
c) Also, as expected, when I run _python weis_driver_loads.py_, I get `None`.

Note that I edited the above and the attached text file to remove my home directory (has my work ID in it) and replaced it with `[home dir]`.

I have attached some output from both running with mpi ("mpi") and without mpi ("nompi").

[example02_output_mpi.txt](https://github.com/user-attachments/files/20618775/example02_output_mpi.txt)
[example02_output_nompi.txt](https://github.com/user-attachments/files/20618776/example02_output_nompi.txt)

What strikes me is that if one diffs the files, the outputs with and without mpi are very similar for most of the run. There appears to be some evidence of multi-threading initially and then none when OpenFAST is executed. This is in contrast with if I run example 1, run_general.py I see interleaved OpenFAST output from both MPI workers, e.g.,

> Generator speed:    2.3 RPM, Pitch angle:   3.4 deg, Power:     0.0 kW, Est. wind Speed:   2.9 m/s
>  Time: 10 of 200 seconds.  Estimated final completion at 18:27:51 (in 0.002 days).                Generator speed:    2.3 RPM, Pitch angle:   3.4 deg, Power:     0.0 kW, Est. wind Speed:   2.9 m/s
>  Time: 10 of 200 seconds.  Estimated final completion at 18:27:46 (in 0.002 days).                Generator speed:    2.7 RPM, Pitch angle:   3.4 deg, Power:     0.0 kW, Est. wind Speed:   2.8 m/s
>  Time: 20 of 200 seconds.  Estimated final completion at 18:27:44 (in 0.002 days).                Generator speed:    2.7 RPM, Pitch angle:   3.4 deg, Power:     0.0 kW, Est. wind Speed:   2.8 m/s
>  Time: 20 of 200 seconds.  Estimated final completion at 18:27:43 (in 0.001 days).                Generator speed:    2.9 RPM, Pitch angle:   3.4 deg, Power:     0.0 kW, Est. wind Speed:   2.9 m/s
>  Time: 30 of 200 seconds.  Estimated final completion at 18:27:44 (in 0.001 days).                Generator speed:    2.9 RPM, Pitch angle:   3.4 deg, Power:     0.0 kW, Est. wind Speed:   2.9 m/s
>  Time: 30 of 200 seconds.  Estimated final completion at 18:27:45 (in 0.001 days).                Generator speed:    3.2 RPM, Pitch angle:   3.4 deg, Power:     0.0 kW, Est. wind Speed:   2.9 m/s

Many thanks! Again much appreciate your insights!
-Brian