NLRWindSystems/WEIS #395: Somethign wrong with example 06_IEA-15-240-RWT weis_driver_umaine_semi.py

maucollu, 2025-05-20:
## Description
I wanted to run the WEIS case weis_driver_umaine_semi.py.
I run it as it is, and it worked, but the OpenFAST simulation is only 1 second long.
Then, in weis_driver_umaine_semi.py, I set TEST_RUN = False, and re-run weis_driver_umaine_semi.py. All ok again, but still 1 second long.
I therefore set the simulation length to 60 seconds as follow:
``

### Steps to reproduce issue
_Please provide a minimum working example (MWE) if possible_

1. Installed WEIS (main)
2. Went to the folder \examples\06_IEA-15-240-RWT
3. modified weis_driver_umaine_semi, setting `TEST_RUN = False`
4. modified modeling_options_umaine_semi.yaml, setting `analysis_time: 60.`
5. run the weis case `python weis_driver_umaine_semi.py`

### Current behavior
When running OpenFAST, the following happens:
'
Generator speed:    7.5 RPM, Pitch angle:   5.9 deg, Power:     0.0 kW, Est. wind Speed:  11.5 m/s

 FAST_Solution:FAST_UpdateStates:FAST_AdvanceStates:SrvD_UpdateStates:DLL_controller_call:BladedInt
 erface option was designed for an explicit-loose coupling scheme. Using last calculated values
 from DLL on all subsequent calls until time is advanced. Warning will not be displayed again.

 FAST_Solution:FAST_UpdateStates:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcO
 utput:BEMT_CalcOutput(node 3, blade 3):UA_CalcOutput:UA_BlendSteady:Temporarily turning off UA
 due to high angle of attack or low relative velocity. This warning will not be repeated though
 the condition may persist.

 FAST_Solution:FAST_UpdateStates:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcO
 utput:BEMT_CalcOutput(node 56, blade 1):UA_CalcOutput:Mach number exceeds 0.3. Theory is invalid.
 This warning will not be repeated though the condition may persist.

 Warning: SkewedWakeCorrection encountered a large value of chi (91.805 deg), so the yaw
 correction will be limited. This warning will not be repeated though the condition may persist.
 See the AD chi output channels, and consider turning off the Pitt/Peters skew model (set
 SkewMod=1) if this condition persists.

 The BEM solution is being turned off due to low TSR.  (TSR = 0.94435). This warning will not be
 repeated though the condition may persist. (See GeomPhi output channel.)

 FAST_Solution:FAST_UpdateStates:FAST_AdvanceStates:SolveOption2a_Inp2BD:ED_CalcContStateDeriv:SetC
 oordSy:Small angle assumption violated in SUBROUTINE SmllRotTrans() due to a large blade
 deflection (ElastoDyn SetCoordSy). The solution may be inaccurate. Simulation continuing, but
 future warnings from SmllRotTrans() will be suppressed.
  Additional debugging message from SUBROUTINE SmllRotTrans(): 2.67 s
 ED_CalcContStateDeriv:SetCoordSy:Small angle assumption violated in SUBROUTINE SmllRotTrans() due
 to a large blade deflection (ElastoDyn SetCoordSy). The solution may be inaccurate. Simulation
 continuing, but future warnings from SmllRotTrans() will be suppressed.
  Additional debugging message from SUBROUTINE SmllRotTrans(): 2.67 s
 FAST_AdvanceStates:AD_UpdateStates:BEMT_UpdateStates:UpdatePhi(node 18, blade
 2):BEMT_UnCoupledSolve:There is no valid value of phi for these operating conditions: Vx =
 41.514, Vy = -608.72, rlocal = 37.674, theta = 0.1752, geometric phi = 3.0735. This warning will
 not be repeated though the condition may persist. (See GeomPhi output channel.)
 FAST_UpdateStates:CalcOutputs_And_SolveForInputs:SolveOption2:AD_CalcOutput:RotCalcOutput:BEMT_Cal
 cOutput(node 3, blade 1):UA_CalcOutput:Mach number exceeds 1.0. Equations cannot be evaluated.

  OpenFAST encountered an error at simulation time 2.67 of 60 seconds.
  Simulation error level: FATAL ERROR

  Aborting OpenFAST.
'

### Expected behavior
OpenFAST running smoothly up to 60 seconds.

### Code versions
- I downgraded openfast-io to 4.0.2 as per other issue

dzalkind, 2025-05-20:
Hi Maurizio,

Does it work when you disable the 2nd tower mode (in both the FA and SS directions)?  https://github.com/WISDEM/WEIS/blob/b9d9d6ee8314530bef542e9b84d79696ccb50a69/examples/06_IEA-15-240-RWT/modeling_options_umaine_semi.yaml#L66

If so, this might be our favorite mode identification issue, @gbarter.

Best, Dan

mcollu, 2025-05-21:
@dzalkind , it is!
I spent a couple of days banging my head on what could it be, even forcing WEIS to have exactly the same control parameters by passing everything in DISCON, and it never worked.
With TwFADOF2 : False and TwSSDOF2 : False, it works!