NLRWindSystems/WEIS #398: Automatic generation of TurbSim bts files with UsableTime "ALL"

mcollu, 2025-06-02:
# Description of feature
Hi, I was wondering if the following option is available and, if not, if it could be implemented.

I need to run 1-hour long OpenFAST simulations (simulated time), for DLC 1.6.
WEIS automatically generates bts files with 1-hour worth of wind speeds.

What I would like instead is the generation of 600 seconds bts files, with the TurbSim input option:
`"ALL"         UsableTime  `

# Potential solution
In [WEIS modeling schema](https://weis.readthedocs.io/en/latest/inputs/modeling_schema.html) the UsableTime can be set to all as:

`    DLCs:
      - DLC: '1.6'
        analysis_time: 3600   #600
        transient_time: 600
        turbulent_wind:
            UsableTime: "ALL" `

but there does not seem to be an option to set up `AnalysisTime`.

I tried, as per the following, but it is ignored, taking as AnalysisTime the "analysis_time" of the DLC

'    DLCs:
      - DLC: '1.6'   # With the new WEIS release, to be substituted by "Steady"
        analysis_time: 3600   #600
        transient_time: 600
        turbulent_wind:
            AnalysisTime: 600
            UsableTime: ALL '

dzalkind, 2025-06-05:
Hi Maurizio,

I believe `UsableTime` is `ALL` by default, so no need for that option.

If you don't want any transient time, `transient_time: 0` is possible.

`AnalysisTime` in turbsim will be `analysis_time + transient_time` for that case.

Please let us know if you need anything clarified further.

Best, Dan

dzalkind, 2025-06-05:
Hi Maurizio,

I'm sure that is possible with a little development effort, but I'm not sure it's possible by only changing the input files at this time.  I'll think about how that could be supported and keep you posted.

Best, Dan

mcollu, 2025-06-05:
Thanks, @dzalkind 

Sure, no problem.
The reason I am asking is because is explicitly mentioned in the IEC-61400-3-x standards as acceptable, and it would save a bit of space generating 10min rather than 1hour or 3hours TurbSim time signals.

dzalkind, 2025-06-05:
Ah, I see.  Our typical approach is to set `n_seeds: 6` (or 18) and let the parallelization of WEIS generate 6 turbsim files.

```
DLC_driver:
    DLCs:
        - DLC: "1.1"
          ws_bin_size: 10
          n_seeds: 6
          analysis_time: 0.1
          transient_time: 0.
```

mcollu, 2025-06-13:
Not sure I understood.
If I have an `analysis_time: 3600`, and I specify `n_seeds: 6`, automatically WEIS will generate 6 3600s turbsim files or 6 600s turbsim files?