NLRWindSystems/WEIS #399: Inquiry Regarding Viscous Damping Force Modeling for Floating Structures in WEIS

fangjianju, 2025-06-05:
Dear WEIS Development Team,

I am writing to inquire about the treatment of viscous damping forces for floating substructures in WEIS. Taking the IEA 15 MW reference turbine as an example, when performing OpenFAST simulations, it is necessary to incorporate viscous damping forces of the floating platform. These forces are typically determined through model tests or CFD simulations.

My question is: How does WEIS efficiently determine the viscous damping forces during floating structure optimization design? Could you please clarify:

What is the current methodology used in WEIS to address this?

Or is the viscous damping effect currently neglected in the optimization process?

Thank you for your insights and expertise on this matter.

Sincerely
Fang

mattEhall, 2025-06-06:
Hi Fang,
I may be giving a partial answer that you already know (apologies if so). Both OpenFAST and RAFT include support for strip-theory hydrodynamics that can include distributed viscous drag/damping forces on the substructure. This works with drag coefficients that can be set for the transverse and axial directions of each member. This is the most automatic way to include viscous drag in that it can adapt with resizing of the substructure geometry and doesn't require prior simulation to determine an overall drag coefficient for the structure as a whole. That being said, it still relies on choosing appropriate drag coefficients for the members, which can be a research topic in itself depending on how much accuracy you need. I typically use pretty standard coefficient values found in literature, but some of my colleagues have looked into this more deeply.
Best,
Matt

dzalkind, 2025-07-18:
Hi Fang,

Sorry for the delay.  The short answer is yes, and more details can be found here: https://weis.readthedocs.io/en/latest/weis_models.html#bem-modeling

Please let us know if you have further questions or specific configurations that we can better support.

Best, Dan