NLRWindSystems/WEIS #400: DLC1.6: multiple wind and wave directions by default

mcollu, 2025-06-06:
# Description of feature
Hi, when using WEIS to run OpenFAST simulations with DLC1.6, on simulations with wind and wave directions equal to zero are run (for all the eligible wind speed, if a wind speed vector is not specified).

But, according to IEC61400-3-2 (2025), DLC1.6 is MIS/MUL, i.e., MULtiple wave directions should be considered, and MISaligned wind/wave directions should be considered.

# Potential solution
In the short term, can you please let me know how can I run different DLC1.6 changing wind/wave directions? I tried specifying in the DLC `wind_heading` and `wave_heading`, but while for other DLCs this work, for this one it seems to be overwritten.

In the longer term, by default, it would be nice that the DLC1.6 would automatically consider, for each wind speed, multiple wind and wave directions.

mcollu, 2025-06-07:
Thanks a lot, @dzalkind.

I tried the following:
`    openfast_input_map:
        wave_dir: [SeaState,WaveDir]
        wind_dir: 
            - [ElastoDyn,NacYaw]
            - [ServoDyn,YawNeut]
        prop_dir:
            - [InflowWind,PropagationDir]
    DLCs:
      - DLC: '6.1'
        label: '6.1'
        wind_speed: [50]
        user_group:
          - wave_dir: [-90,0,90]
          - wind_dir: [-90,0,90]
          - prop_dir: [90,0,-90]  # sign of propagation direction is opposite to nacelle heading in OpenFAST
        analysis_time: 3600   #3600
        transient_time: 600 #600
        turbulent_wind:
            AnalysisTime: 600
            UsableTime: ALL
        # wind_heading: [-90]
        wave_heading: [0]
        pitch_initial: [2.426047, 2.426047, 0.377375, 0.000535, 0.000535, 1.170321,
            6.052129, 9.189114, 11.824437, 14.19975, 16.42107, 18.525951, 20.553121,
            20.553121, 20.553121]
        rot_speed_initial: [5.000012, 5.000012, 5.000012, 5.000012, 6.390847, 7.559987,
            7.559987, 7.559987, 7.559987, 7.559987, 7.559987, 7.559987, 7.559987,
            7.559987, 7.559987]`

And it correctly generate 27 OpenFAST tests in the test matrix, i.e. 3 wave dire x 3 wind dir x 3 nacelle positions.

But then, I got an error that is not trivial to me...cannot understand what it mean:

`Traceback (most recent call last):
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\openmdao\core\system.py", line 2931, in _call_user_function
    yield
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\openmdao\core\explicitcomponent.py", line 278, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "C:\Users\mauri\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 670, in compute
    case_list, case_name, dlc_generator  = self.run_FAST(inputs, discrete_inputs, fst_vt)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\mauri\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 2034, in run_FAST
    case_df.insert(0,'DLC',dlc_label)
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\pandas\core\frame.py", line 5171, in insert
    value, refs = self._sanitize_column(value)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\pandas\core\frame.py", line 5266, in _sanitize_column
    com.require_length_match(value, self.index)
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\pandas\core\common.py", line 573, in require_length_match
    raise ValueError(
ValueError: Length of values (54) does not match length of index (27)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\mauri\OneDrive - University of Strathclyde\NUM_FMS21289\scube\tests\test_serial.py", line 65, in <module>
    wt_opt, modeling_options, opt_options = weis_main(fname_wt_input,
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\mauri\WEIS\weis\main.py", line 110, in weis_main
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input,
                                            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\mauri\WEIS\weis\glue_code\runWEIS.py", line 187, in run_weis
    wt_opt.run_model()
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\openmdao\core\problem.py", line 689, in run_model
    self.model.run_solve_nonlinear()
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\openmdao\core\system.py", line 4963, in run_solve_nonlinear
    self._solve_nonlinear()
[... 10 lines cut ...]
    self._compute_wrapper()
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\openmdao\core\explicitcomponent.py", line 261, in _compute_wrapper
    with self._call_user_function('compute'):
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\contextlib.py", line 158, in __exit__
    self.gen.throw(value)
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\openmdao\core\system.py", line 2937, in _call_user_function
    raise err_type(
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\openmdao\core\system.py", line 2931, in _call_user_function
    yield
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\openmdao\core\explicitcomponent.py", line 278, in _compute_wrapper
    self.compute(self._inputs, self._outputs,
  File "C:\Users\mauri\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 670, in compute
    case_list, case_name, dlc_generator  = self.run_FAST(inputs, discrete_inputs, fst_vt)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\mauri\WEIS\weis\aeroelasticse\openmdao_openfast.py", line 2034, in run_FAST
    case_df.insert(0,'DLC',dlc_label)
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\pandas\core\frame.py", line 5171, in insert
    value, refs = self._sanitize_column(value)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\pandas\core\frame.py", line 5266, in _sanitize_column
    com.require_length_match(value, self.index)
  File "C:\Users\mauri\miniforge3\envs\weis-env2\Lib\site-packages\pandas\core\common.py", line 573, in require_length_match
    raise ValueError(
ValueError: 'aeroelastic' <class FASTLoadCases>: Error calling compute(), Length of values (54) does not match length of index (27)`

mcollu, 2025-06-07:
It may be a case, but I noticed that 54 = 2 * 27, and I noticed that when wind direction / nac yaw is not specified, it automatically misalign the nacelle +-8 deg. It may be that the algorithm, for each one of my cases, expect a -8+8 misalignment?

mcollu, 2025-06-11:
No need to thank me, @dzalkind , I am very grateful for the hard work you've all put into WEIS and the other tools.

In the following, I thought there is a typo, i.e., a missing dash "-":
```
          user_group:
            - wave_dir: [-90,0,90]
            - wind_dir: [-90,0,90]
              prop_dir: [90,0,-90]  # sign of propagation direction is opposite to nacelle heading in OpenFAST
```

But the dash is not missing, is it? It means that, without dash, the wind_dir -90 is always coupled with prop_dir+90, the 0 with 0, and the 90 with -90, right?
I think I now understand why you said I may have downwind wind turbines...and some simulations failing!
Instead, if I put a "-", WEIS automatically does all the combinations.

Many thanks for the clarification about yaw_misalign - I've imposed `yaw_misalign: [0]` and it now works!

dzalkind, 2025-06-11:
> But the dash is not missing, is it? It means that, without dash, the wind_dir -90 is always coupled with prop_dir+90, the 0 with 0, and the 90 with -90, right?

Your understanding on the dash and coupling is correct.  Glad it's working!

mcollu, 2025-06-11:
Many thanks, I'll close this thread then.