NLRWindSystems/WEIS #406: 03_design_with_openfast errors

bluemiracle4862, 2025-07-04:
I got the following error in executing ex.3 from develop branch.

### Current behavior

    raise RuntimeError('\n'.join(final_msg))
RuntimeError: 
Collected errors for problem 'iea22_ptfm_opt_driver':
   <model> <class WindPark>: Attempted to connect from 'rotorse.re.generate_KI.K' to 'aeroelastic.blade:K', but 'rotorse.re.generate_KI.K' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['rotorse.re.edge_iner', 'rotorse.re.Tw_iner', 'rotorse.re.K'].
   <model> <class WindPark>: Attempted to connect from 'rotorse.re.generate_KI.I' to 'aeroelastic.blade:I', but 'rotorse.re.generate_KI.I' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['rotorse.re.edge_iner', 'rotorse.rs.frame.EI22', 'rotorse.rs.frame.EI11'].
   <model> <class WindPark>: Attempted to connect from 'nacelle.L_generator' to 'drivese_post.L_generator', but 'nacelle.L_generator' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['nacelle.gear_ratio', 'generator.L_generator', 'drivese.s_generator'].

could you solve this error ?

very thanks.

gbarter, 2025-07-07:
Please make sure your WISDEM package is up-to-date.  If you update WISDEM and the error still occurs, please let us know.

bluemiracle4862, 2025-07-10:
I updated WISDEM to version 3.22.0.
But I am getting the same error.

---------------------------------------------------------------------------------------------------------
DeprecationWarning: /home/m/anaconda3/envs/WEIS-develop-2571/lib/python3.12/site-packages/jsonmerge/__init__.py:27
Accessing Draft4Validator.resolver is deprecated as of v4.18.0, in favor of the https://github.com/python-jsonschema/referencing library, which provides more compliant referencing behavior as well as more flexible APIs for customization.PendingDeprecationWarning: /home/m/WEIS-develop/weis/dtqpy/classes/DTQPy_CLASS_INTERNAL.py:10
Importing from numpy.matlib is deprecated since 1.19.0. The matrix subclass is not the recommended way to represent matrices or deal with linear algebra (see https://docs.scipy.org/doc/numpy/user/numpy-for-matlab-users.html). Please adjust your code to use regular ndarray. DeprecationWarning: /home/m/anaconda3/envs/WEIS-develop-2571/lib/python3.12/site-packages/jsonmerge/__init__.py:27
Accessing Draft4Validator.resolver is deprecated as of v4.18.0, in favor of the https://github.com/python-jsonschema/referencing library, which provides more compliant referencing behavior as well as more flexible APIs for customization.Traceback (most recent call last):
  File "/home/m/WEIS-develop/examples/03_design_with_openfast/iea22_ptfm_opt_driver.py", line 14, in <module>
    wt_opt, modeling_options, opt_options = weis_main(fname_wt_input, 
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m/WEIS-develop/weis/main.py", line 110, in weis_main
    wt_opt, modeling_options, opt_options = run_weis(fname_wt_input, 
                                            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m/WEIS-develop/weis/glue_code/runWEIS.py", line 185, in run_weis
    wt_opt.run_driver()
  File "/home/m/anaconda3/envs/WEIS-develop-2571/lib/python3.12/site-packages/openmdao/core/problem.py", line 721, in run_driver
    self.final_setup()
  File "/home/m/anaconda3/envs/WEIS-develop-2571/lib/python3.12/site-packages/openmdao/core/problem.py", line 1075, in final_setup
    self._check_collected_errors()
  File "/home/m/anaconda3/envs/WEIS-develop-2571/lib/python3.12/site-packages/openmdao/core/problem.py", line 626, in _check_collected_errors
    raise RuntimeError('\n'.join(final_msg))
RuntimeError: 
Collected errors for problem 'iea22_ptfm_opt_driver':
   <model> <class WindPark>: Attempted to connect from 'rotorse.re.generate_KI.K' to 'aeroelastic.blade:K', but 'rotorse.re.generate_KI.K' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['rotorse.re.edge_iner', 'rotorse.re.Tw_iner', 'rotorse.re.K'].
   <model> <class WindPark>: Attempted to connect from 'rotorse.re.generate_KI.I' to 'aeroelastic.blade:I', but 'rotorse.re.generate_KI.I' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['rotorse.re.edge_iner', 'rotorse.rs.frame.EI22', 'rotorse.rs.frame.EI11'].
   <model> <class WindPark>: Attempted to connect from 'nacelle.L_generator' to 'drivese_post.L_generator', but 'nacelle.L_generator' doesn't exist. Perhaps you meant to connect to one of the following outputs: ['nacelle.gear_ratio', 'generator.L_generator', 'drivese.s_generator'].

Process finished with exit code 1
Please review and advise it.

gbarter, 2025-07-10:
Sorry about that.  Can you try the `develop` branch of WEIS?  WEIS is likely to push a new release soon.

dzalkind, 2025-07-18:
This should be fixed in v1.6.  Please let us know if you still have issues.

closed 2025-07-18 by dzalkind