NLRWindSystems/WEIS #411: 17_IEA22_openfast_runs

Wying301, 2025-07-18:
Hi, I cannot run the openfast simulations (no optimization) in WEIS by loading IEA22 FAST model downloaded here https://github.com/IEAWindSystems/IEA-22-280-RWT/tree/main/OpenFAST/IEA-22-280-RWT-Semi. 

The issue is there are some changes in HydroDyn.dat, using CYLINDRICAL MEMBER and RECTANGULAR MEMBER properties rather than MEMBER properties. And some parameters are the same, such as line 90-107 in IEA-22-280-RWT-Semi_HydroDyn_PotMod.dat file. As shown below, "PropThck" appears in both parts, which means the second will cover the first when reading this file and leads to error. 

---------------------- CYLINDRICAL MEMBER CROSS-SECTION PROPERTIES -------------------------
10          NPropSetsCyl   - Number of member property sets (-)
 PropSetID     PropD     **PropThck**  
    (-)         (m)         (m)    
     1         12.0        0.05    ! Center col    
     2         12.5        0.05    ! Upwind col    
     3         12.5        0.05    ! Stbd col    
     4         12.5        0.05    ! Port col    
     5         0.91        0.02    ! Top brace - upwind    
     6         0.91        0.02    ! Top brace - stbd    
     7         0.91        0.02    ! Top brace - port    
     8         10.0        0.04    ! Pontoon - upwind    
     9         10.0        0.04    ! Pontoon - stbd    
    10         10.0        0.04    ! Pontoon - port    
---------------- RECTANGULAR MEMBER CROSS-SECTION PROPERTIES -------------------
             0   NPropSetsRec    - Number of rectangular member property sets (-)
MPropSetID   PropA      PropB    **PropThck**
   (-)        (m)        (m)       (m)

Thanks for any help!

dzalkind, 2025-07-18:
The latest version of WEIS should support OpenFAST v4.1.  This is the updated example for running your own IEA-22MW model: https://github.com/WISDEM/WEIS/blob/main/examples/01_simulate_own_openfast_model/dlc_sim_driver.py

dzalkind, 2025-07-24:
Yes, that script will run batch cases of OpenFAST.  WEIS uses the same methods.  The example is configured to run steady state wind conditions.  Here is a more detailed account of the examples: https://weis.readthedocs.io/en/latest/weis_examples.html#simulate-own-openfast-model