NLRWindSystems/WEIS #417: DOE with MPI never ends

yonghoonlee, 2025-08-01:
## Description
Design of experiment using MPI never ends - OpenMDAO stuck at finalizing the DOE run and never completes the process.
The functionality worked until OF 4.1 implementation.
Before July 2 commits, the DOE works fine.
Once I merge July 2 commits (I guess they are mostly related to OF 4.1), the functionality breaks down.

Note: Serial run without MPI works, and looks fine.
After July 2 commits, it shows more warning messages, though.
(Example: `sys:1: DeprecationWarning: builtin type swigvarlink has no __module__ attribute`)

### Steps to reproduce issue
1. Change umain_semi RAFT example by turning optimization off and design_of_experiment on.
2. Run driver using mpiexec

### Current behavior
Process never ends.

### Expected behavior
Process should end.

### Code versions
_List versions only if relevant_
- Python
- …

dzalkind, 2025-08-01:
Hi Yong Hoon,

I'm seeing this, too.  It's odd because I don't see any of those commits affecting the parts of WEIS that deal with MPI:
- https://github.com/WISDEM/WEIS/blob/main/weis/main.py
- https://github.com/WISDEM/WEIS/blob/main/weis/glue_code/mpi_tools.py (this is where I'd start)
- https://github.com/WISDEM/WEIS/blob/main/weis/glue_code/runWEIS.py

My Friday afternoon guess is that we're spinning up the wrong number of cores/ranks and WEIS is sitting around waiting for one that is not doing anything.  We worked hard on the MPI set up for optimizations that use finite differencing, but the DOE set up may be less matured.

I hope this helps.  I will check in next week.

Best, Dan

ptrbortolotti, 2025-08-03:
hi both, I've spent a couple of days debugging this, unfortunately without success. I am 99% sure that it is an issue with some underlying library. However not with `openmdao` (see https://github.com/OpenMDAO/OpenMDAO/issues/3571), nor `mpi4py` (https://github.com/mpi4py/mpi4py/issues/669 and https://github.com/mpi4py/mpi4py/issues/671)

dzalkind, 2025-08-04:
Hi @yonghoonlee, I tracked this issue down to this commit: https://github.com/WISDEM/WISDEM/commit/c217c72700c7677f3acef5427063afa88e0cfce8

If you comment out or remove those lines, it completed for me.  

I don't fully understand why that `Barrier()` function was causing an issue.

yonghoonlee, 2025-08-04:
Thank you @dzalkind and @ptrbortolotti for taking time to investigate on it.

@dzalkind I didn't test yet, but based on the information you provided, I think I got to know why this issue is occurring. I called `save_yaml` logic from `rank=0` only, as I thought calling that from all the MPI cores does not make sense. However, because the `MPI.COMM_WORLD.Barrier()` is called inside the `write_yaml` logic, it seems that I need to call the function from all MPI cores to pass that barrier.

I personally do not think that it is natural to call yaml writing function from all cores, as only rank=0 would write the yaml file and all other cores will do nothing. So, having MPI Barrier inside that logic would induce similar mistakes for developers (including me) who do not exactly know the `write_yaml` function structure. But, I can modify my code to call it from all cores, if NREL decide to keep the MPI Barrier inside that logic. Thank you.